arendst/Tasmota #5425: S20, 6.4.1 KNX function stops after a while. Device has to be restarted

MachmaHinne, 2019-03-09:
### BUG DESCRIPTION
S20 device is configured, works for a few hours and then the KNX function stops. Web interface is still working, can be switched on and off. Also the button is still working.
After rebooting KNX is working again.
Device is bound via a raspi with newest knxd and via an ip interface to the bus.

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : S20
- [x] Tasmota binary firmware version number used : 6.4.1 precompiled with knx 
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [ x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
01:21:42 RSL: STATUS = {"Status":{"Module":8,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
01:21:42 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-knx.bin","RestartReason":"Software/System restart","Uptime":"0T00:45:32","StartupUTC":"2019-03-08T23:36:10","Sleep":50,"BootCount":12,"SaveCount":294,"SaveAddress":"F6000"}}
01:21:42 RSL: STATUS2 = {"StatusFWR":{"Version":"6.4.1(knx)","BuildDateTime":"2018-12-24T14:49:38","Boot":6,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
01:21:42 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["gnmpf",""],"TelePeriod":300,"SetOption":["0A008001","558180C0","00000001"]}}
01:21:42 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":537,"Free":464,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","1FDAE594","000383B0","23B617CE","00003BC0"]}}
01:21:42 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff-6691","IPAddress":"192.168.2.74","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"84:F3:EB:4A:BA:23","Webserver":2,"WifiConfig":4}}
01:21:42 RSL: STATUS7 = {"StatusTIM":{"UTC":"Sat Mar 09 00:21:42 2019","Local":"Sat Mar 09 01:21:42 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:16","Sunset":"18:45"}}
01:21:42 RSL: STATUS10 = {"StatusSNS":{"Time":"2019-03-09T01:21:42"}}
01:21:42 RSL: STATUS11 = {"StatusSTS":{"Time":"2019-03-09T01:21:42","Uptime":"0T00:45:32","Vcc":3.511,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"myssid,"BSSId":"7C:FF:4D:E7:21:E4","Channel":6,"RSSI":100}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
Unfortunatly I restarted it one hour ago, and  I cannot provide it.

```

### TO REPRODUCE
wait for disfunction

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-03-09:
Hi,

I have this feature working at home for more than a year without issues, so we would need to do some tests to find the problem in your setup. So, please be patient with all the tests we have to do. 

First, please update to latest Tasmota and with core 2.3.0. Let's start testing that. Link to latest: http://thehackbox.org/tasmota/020300/sonoff-knx.bin

MachmaHinne, 2019-03-09:
Hi,

Thank you for your quick response. I forgot, that I do have another device in the cellar. Same problem. I switched to Weblog 4:
08:35:30 RSL: STATE = {"Time":"2019-03-09T08:35:30","Uptime":"58T08:36:18","Vcc":3.553,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"gnmpf","BSSId":"7C:FF:4D:F1:87:53","Channel":1,"RSSI":30}}
08:35:33 WIF: Checking connection...

But after sending an on command via knx to it there was no action at the log.
May be it is informativ to send my knxd parameters:
/usr/bin/knxd --eibaddr=1.1.128 --client-addrs=1.1.129:10 -c -f 1 -D -T -R -S --listen-local=/tmp/knx -b ipt:192.168.2.8
After sending 
knxtool on ip:127.0.0.1 9/2/5
The vbusmonitor responded
LPDU: BC 11 87 4A 05 E1 00 81 FA :L_Data low from 1.1.135 to 9/2/5 hops: 06 T_DATA_XXX_REQ A_GroupValue_Write (small) 01
but no response:
[image]

And now I will follow your advice with changing the firmware.

MachmaHinne, 2019-03-09:
After coming home now, I tried. The failure has occured again. :-(
Checked the log. Only for 25min is available. Nothing obvious to see.

MachmaHinne, 2019-03-09:
Ok. Now, a few hours later is working again, without a restart. Strange.

ascillato, 2019-03-11:
Hi, 

This issue have happened before: https://github.com/ascillato/Sonoff-Tasmota_KNX/issues/82 and also in https://github.com/arendst/Sonoff-Tasmota/issues/2643

This lost of multicast communication (as you can see in the referenced issues) was because a problem in the router or wifi setup. The IGMP used in the routers for multicast, don't work well if you use a wifi repeater or some brands of wifi routers due the esp8266 arduino core is not 100% compatible with all brands. 

Can you access to your wifi router configuration to see:
- If multicast is turned on ?
- IGMP is v3 ?
- Multicast Snoop is off ?

As explained in https://github.com/arendst/Sonoff-Tasmota/issues/2643#issuecomment-388763990 the solution was connecting Tasmota directly to the main router, avoiding repeaters and avoiding fritzbox.

Please, tell us how is your wifi setup, your wifi config, if there are repeaters, if you can connect directly to the router, if you have fritzbox. Thanks

MachmaHinne, 2019-03-11:
Hi,

Thanks again for your help. I understand the problem partially.
To set up the work around will be difficult for me, because I´m using really a fritzbox with this new "mesh"-technology. I bound three other fritzboxes via a cabled backbone to the main router.
Multicast v3 is on by default, but there is no option for the snooping.
Avoiding the mesh repeaters will not be possible, because my sonoffs are distibuted at a lot of places in the building. 

So I will study the other issues carefully and try to find a workaround.

Greetings
Peter

ascillato, 2019-03-12:
Fritzbox and mesh are problematic for the arduino core.

You can try Tasmota with latest core 2.5.0 that have a some bugs fixed about mesh support. Beyond that there is much we can do because it is outside Tasmota and it is in the arduino core.

ascillato, 2019-03-12:
http://thehackbox.org/tasmota/020500/

MachmaHinne, 2019-03-12:
Ok. I´ll try it. Thank you for all your patience.

MachmaHinne, 2019-03-13:
Log:
working for 1 day without failure.

ascillato2, 2019-03-14:
Great to know that now it is working for you. Thanks for all the testings :+1:

closed 2019-03-14 by ascillato2

MachmaHinne, 2019-03-16:
Yes, this device is working now for 3 days without failure. That´s fine.
But I have another S20 device flashed also with the 2.5.0 tasmota. It´s connected to another node of my fritzbox mesh - the main router (7490, the mesh repeaters are 4040 bound together by a ethernet backbone) in the cellar - and it´s not receiving any knx telegrams. But the vbusmonitor says:
LPDU: BC 11 85 4A 05 E1 00 81 F8 :L_Data low from 1.1.133 to 9/2/5 hops: 06 T_DATA_XXX_REQ A_GroupValue_Write (small) 01 

I can toggle th S20 by the WebUI and the answers are recognized by the vbusmonitor1:
 LPDU: BC 00 1F 4A 05 E1 00 81 73 :L_Data low from 0.0.31 to 9/2/5 hops: 06 T_DATA_XXX_REQ A_GroupValue_Write (small) 01 
LPDU: BC 00 1F 4A 05 E1 00 80 72 :L_Data low from 0.0.31 to 9/2/5 hops: 06 T_DATA_XXX_REQ A_GroupValue_Write (small) 00 

Very strange.  What was the name of the wlan sniffer app you used?

Greetings 
Peter

ascillato, 2019-03-17:
>What was the name of the wlan sniffer app you used?

Multicast tester

Zaschii, 2024-01-27:
What was the final solution Here?
Link http://thehackbox.org/tasmota/020500/ IS Not valid anymore.

I have exactly the Same issue with Fritzbox Router and unifi Access Points. Direct Connection to fritzbox is no Option. 

Thx and br

barbudor, 2024-01-28:
Don't hijack an old issue that is 5 years old and many things has changed since
Open a new discussion and reports all details based on latest Tasmota version