arendst/Tasmota #5426: Tasmota version 6.4.1 connection issue

gabrielvargars, 2019-03-09:
Hi,

Im a beginner. Im flashed on a brand new sonoff basic a tasmota 6.4.1. the device is not connecting to wifi.
Im already try 3 different networks. 

is it possible to downgrade to lower version like 5?

Thank you

Gabriel

localhost61, 2019-03-09:
Yes, no version control, you can flash what you want. But I don't think that downgrading will provide a solution to your issue. See the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Initial-Configuration) for more help.
You can see what happen if you connect a terminal on the serial port at (115200/8N1 and no flow control). And this interface should provide you a mean to send plenty of [commands](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands) to control the device.

blakadder, 2019-03-09:
try a latest development build from http://thehackbox.org/tasmota/ and try other cores than the default 2.4.2 if you're having wifi issues and didn't misconfigure ssid and password

Jason2866, 2019-03-10:
Before flashing a new version, erase the whole flash with esptool.py erase_flash.
Fragments left in flash can make weired issues....

ascillato2, 2019-03-10:
Closing this issue as it has been answered.

Please, if you need further assistance, find us in the Tasmota Support Chat. Github issues are for bug reporting with the provided template completed. Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-03-10 by ascillato2