arendst/Tasmota #5427: PIR doesn't work after NodeMCU unplugged/ power failure

coldfire84, 2019-03-09:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION
Tested with Tasmota development 6.4.1.20 and latest stable on multiple Node-MCU v1/ Amica boards which have the following sensors attached (GPIO layout in screenshot further down):
* BH1750
* DHT22
* AM312 (PIR)

An LED is also plugged in to use as a visual notification for PIR trigger.

After setting the module to "Generic" and configuring GPIO everything works as expected - sensor data is shown in the Tasmota web interface and MQTT messages are sent/ the LED lights up when the PIR is triggered. 

The issue I have occurs when the Node-MCU board is powered off/ unplugged. On reconnecting power the PIR has a "mind of its own" and does not seem to respond to anything logical. i.e. if I cover the PIR up it still randomly triggers. It stays "ON" or "OFF" for extended periods (5-10 seconds, vs the 2 seconds it should stay on for), ignoring any movement.

I have found that if I simply open the Tasmota GPIO configuration and hit "Save," after the automatic restart, everything starts to work as expected again. Subsequent "warm reboots" work without issue as well, so this issue seems to only occur when power is removed from the device. The issue occurs every time power is removed.

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : [NodeMCU v1/ Amica](https://www.electronicwings.com/public/images/user_images/images/NodeMCU/NodeMCU%20Development%20Boards/Node-MCU-Devkit-1-0.jpg)
- [x] Tasmota binary firmware version number used : 6.4.1.20 (self-compiled)
- [x] Development IDE - Compiler / Upload tools used : PlatformIO
- [x] Provide the output of command ``status 0`` :
```
====================================================================
STATUS 0 **when working**:
====================================================================
11:02:33 MQT: stat/sensor-esp2/STATUS = {"Status":{"Module":18,"FriendlyName":["Study Multisensor"],"Topic":"sensor-esp2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:02:33 MQT: stat/sensor-esp2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sensors","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:05:40","StartupUTC":"2019-03-09T09:56:53","Sleep":50,"CfgHolder":4680,"BootCount":10,"SaveCount":420,"SaveAddress":"F8000"}}
11:02:33 MQT: stat/sensor-esp2/STATUS2 = {"StatusFWR":{"Version":"6.4.1.20(sonoff)","BuildDateTime":"2019-03-08T23:05:15","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
11:02:33 MQT: stat/sensor-esp2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IOT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
11:02:33 MQT: stat/sensor-esp2/STATUS4 = {"StatusMEM":{"ProgramSize":527,"Free":476,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1625C2","FlashMode":3,"Features":["00000809","0FDAE794","240783A1","23B617CE","00003BC0"]}}
11:02:33 MQT: stat/sensor-esp2/STATUS5 = {"StatusNET":{"Hostname":"sensor-esp2-6143","IPAddress":"172.16.2.95","Gateway":"172.16.2.254","Subnetmask":"255.255.255.0","DNSServer":"172.16.2.254","Mac":"xx:xx:xx:xx:xx:xx","Webserver":2,"WifiConfig":5}}
11:02:33 MQT: stat/sensor-esp2/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt-server-dns-name","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_CAF7FF","MqttUser":"user-name","MqttType":1,"MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
11:02:33 MQT: stat/sensor-esp2/STATUS7 = {"StatusTIM":{"UTC":"Sat Mar 09 10:02:33 2019","Local":"Sat Mar 09 11:02:33 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:16","Sunset":"18:45"}}
11:02:33 MQT: stat/sensor-esp2/STATUS10 = {"StatusSNS":{"Time":"2019-03-09T11:02:33","Switch1":"ON","SI7021":{"Temperature":21.8,"Humidity":59.4},"BH1750":{"Illuminance":35},"TempUnit":"C"}}
11:02:33 MQT: stat/sensor-esp2/STATUS11 = {"StatusSTS":{"Time":"2019-03-09T11:02:33","Uptime":"0T00:05:40","Vcc":3.005,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"IOT","BSSId":"xx:xx:xx:xx:xx:xx","Channel":6,"RSSI":46,"LinkCount":1,"Downtime":"0T00:00:04"}}}

====================================================================
STATUS 0 **when broken**:
====================================================================
11:05:31 MQT: stat/sensor-esp2/STATUS = {"Status":{"Module":18,"FriendlyName":["Study Multisensor"],"Topic":"sensor-esp2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:05:31 MQT: stat/sensor-esp2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sensors","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"0T00:00:21","StartupUTC":"2019-03-09T10:05:10","Sleep":50,"CfgHolder":4680,"BootCount":11,"SaveCount":448,"SaveAddress":"F4000"}}
11:05:31 MQT: stat/sensor-esp2/STATUS2 = {"StatusFWR":{"Version":"6.4.1.20(sonoff)","BuildDateTime":"2019-03-08T23:05:15","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
11:05:31 MQT: stat/sensor-esp2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IOT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
11:05:31 MQT: stat/sensor-esp2/STATUS4 = {"StatusMEM":{"ProgramSize":527,"Free":476,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1625C2","FlashMode":3,"Features":["00000809","0FDAE794","240783A1","23B617CE","00003BC0"]}}
11:05:31 MQT: stat/sensor-esp2/STATUS5 = {"StatusNET":{"Hostname":"sensor-esp2-6143","IPAddress":"172.16.2.95","Gateway":"172.16.2.254","Subnetmask":"255.255.255.0","DNSServer":"172.16.2.254","Mac":"xx:xx:xx:xx:xx:xx","Webserver":2,"WifiConfig":5}}
11:05:31 MQT: stat/sensor-esp2/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt-server","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_CAF7FF","MqttUser":"username","MqttType":1,"MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
11:05:31 MQT: stat/sensor-esp2/STATUS7 = {"StatusTIM":{"UTC":"Sat Mar 09 10:05:31 2019","Local":"Sat Mar 09 11:05:31 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:16","Sunset":"18:45"}}
11:05:31 MQT: stat/sensor-esp2/STATUS10 = {"StatusSNS":{"Time":"2019-03-09T11:05:31","Switch1":"ON","SI7021":{"Temperature":22.0,"Humidity":61.5},"BH1750":{"Illuminance":30},"TempUnit":"C"}}
11:05:31 MQT: stat/sensor-esp2/STATUS11 = {"StatusSTS":{"Time":"2019-03-09T11:05:31","Uptime":"0T00:00:21","Vcc":3.002,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"IOT","BSSId":"xx:xx:xx:xx:xx:xx","Channel":6,"RSSI":54,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [x] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
[... cut ...]
When the Node-MCU board is powered off/ unplugged. On power on the PIR has a "mind of its own" and does not seem to respond to anything logical. i.e. if I cover the PIR it still randomly triggers.

If I simply open the Module Configuration and hit "Save"  after the automatic restart everything starts to work as expected again. Subsequent "warm reboots" work without issue as well, so this issue seems to only occur when power is removed from the device.

### EXPECTED BEHAVIOUR
After power off/ unplug the device should power on and the PIR function as expected.

### SCREENSHOTS
GPIO configuration:
[image]

Jason2866, 2019-03-09:
Using GPIO 0 is a bad idea. Try GPIO 3 or 1 instead.

coldfire84, 2019-03-09:
> Using GPIO 0 is a bad idea. Try GPIO 3 or 1 instead.

OK, I can test that, I just followed the GPIO example I in the wiki [here](https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos-D1-Mini-and-BH1750-Luminosity-Sensor)

Jason2866, 2019-03-09:
GPIO 0 can work...

coldfire84, 2019-03-09:
Changed as below, issue persists:

[image]

Jason2866, 2019-03-10:
PIR hardware related than. Search in issues, there are many suggestions to solve....

coldfire84, 2019-03-10:
> PIR hardware related than. Search in issues, there are many suggestions to solve....

How do you figure that? The multi-sensor/ PIR works as expected when GPIO is initially configured. It just stops working when power is removed. I've also used the Bruh Automation code and it works without issue. I'd just much rather use Tasmota as-per the rest of the devices I have.

Jason2866, 2019-03-10:
There are many different issues with PIR sensors. IF low and high levels are correct it will work fine.
Why i can say this? I have PIR sensors installed and all work fine (yes, after a power cycle too).

coldfire84, 2019-03-10:
Resolved, changed SwitchMode to 1.

closed 2019-03-10 by coldfire84