arendst/Tasmota #5428: Heater feature "Sonoff Si7021 sensor"

Georgex12, 2019-03-09:
Is there the possibility to integrate in Tasmota a function to  turn on/off the heater feature of an "Sonoff Si7021 sensor" ?  As soon as the humidity readings are >80% and/or there is condensation on the sensor (which happens if the sensor is mounted outside...) readings get crazy and very unreliable. I know that the "Sonoff Si7021 sensor" is forced by an onboard MCU to commnicate as 1-wire device...

Jason2866, 2019-03-10:
Yes, take a look in wiki under section rules and the examples

curzon01, 2019-03-10:
afaik Si7021 heater currently not supported. In case doing so it would be also difficult to get correct temperature values (as shown [here](https://idyl.io/arduino/how-to/interface-si7021-sensor-arduino/#attachment_705)). Also then it must be possible to set the desired heater current (3 mA up to 93 mA)

ascillato2, 2019-03-10:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-03-10 by ascillato2

duncanbarnes, 2021-11-13:
> Are you all aware that this topic here is about the Sonoff version of the Si7021 ? Where the SI7021 is encapsulated by Sonoff with a MCU to simulate the 1-wire READ-ONLY protocol of AM2301 ? There is no way to tell the sensor to stop heating apart from totally cutting down it's power. End of the story : if you want the Sonoff SI7021 to stop heating, power it through a GPIO as a RELAY and shut it down. The feature is already supported !

Just to be clear, you are suggesting that the heating should be working? Its described as unsupported [on the wiki](https://tasmota.github.io/docs/devices/Sonoff-TH/) suggesting it does not. My own sensor situated in a shower room was reading 18% after I had a shower this morning which also suggests it is not working and getting confused by condensation on the sensor. I'd certainly love to see this working but it certainly doesn't under Tasmota at the moment.

Are you also saying that the Sonoff electronic design does not have the ability to control the sensor? Again the wording on the wiki suggests its possible.

barbudor, 2021-11-13:
The wording on the documentation clearly states that it is not possible to control the heating from the ESP.
This is independant of running Sonoff firmware or Tasmota on the ESP.
Again and again : The so-called "Sonoff SI7021" is an I2C Si7021 + a local MCU.
The SI7021 is controlled by the MCU and the MCU sends measurments ONE-WAY to the ESP on a pseudo 1-wire protocol that mimics the DHT protocol.
So if Sonoff didn't enabled the heater in their MCU, there is no way to enable it or disable it at will.

If you want serious thing, move to I2C

duncanbarnes, 2021-11-13:
Ok, the two reasons I questioned it were, firstly the initial responses to this questioned how it would work in connection to the temperature sensor which seemed strange to be discussing if there was a hardware limitation that made such a conversation irrelevant. Secondly, comments on this issue include "currently unsupported", and some of the wording in the wiki all suggested a similar ambiguity, so possible, just not now. But good to know, will look at other options. Perhaps the AM2301, can't see a clear answer on how that interfaces but some suggestions online that it performs better.