arendst/Tasmota #5429: Sonoff 1-channel reboot switches problem

badigit, 2019-03-09:
### ISSUE DESCRIPTION - TROUBLESHOOTING
Have sonoff 1-channel switch, look like at photo
[image]
If the current relay state is OFF and im trying restart devicevia web ui, or terminal (command "restart 1") relay makes switches 1-0. It not shows on terminal log.

If the current state is ON, then, when trying soft restart, relay no switching

already tried: flash several versions tasmota (precompiled\selfcompiled), incl last dev build.
Trying full erase ESP8255 firmware, several times - it doesnt help.

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Sonoff 1-Channel inching
- [x] Tasmota binary firmware version number used : 6.4.1.20 / (tryed pre-compiled AND self-compiled)
- [x] Development IDE - Compiler / Upload tools used : VisualStudioCode / Platformio, esptool, ESPEasyFlasher
- [x] Provide the output of command ``status 0`` : 
```
14:59:07 CMD: status 0
14:59:07 RSL: stat/sonoff/STATUS = {"Status":{"Module":12,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:59:07 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:12","StartupUTC":"2019-03-09T13:58:55","Sleep":50,"CfgHolder":4617,"BootCount":9,"SaveCount":22,"SaveAddress":"F5000"}}
14:59:07 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1.20(classic)","BuildDateTime":"2019-03-09T16:18:41","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
14:59:07 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["DD","indebuurt1"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
14:59:07 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":499,"Free":504,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","6010A790","248403A8","210217C6","000000C0"]}}
14:59:07 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-0725","IPAddress":"192.168.2.236","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"60:01:94:88:C2:D5","Webserver":2,"WifiConfig":4}}
14:59:07 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_88C2D5","MqttUser":"DVES_USER","MqttType":1,"MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
14:59:07 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sat Mar 09 13:59:07 2019","Local":"Sat Mar 09 14:59:07 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00"}}
14:59:07 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-03-09T14:59:07"}}
14:59:07 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-03-09T14:59:07","Uptime":"0T00:00:12","Vcc":3.475,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"DD","BSSId":"30:5A:3A:5C:A2:A8","Channel":9,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```

Jason2866, 2019-03-10:
This is nomal. It depends which GPIO pin is used to control the relay. Most of GPIO pins change there state through a power cycle. This is by design from the ESP chip an not under the control of Tasmota

badigit, 2019-03-10:
This not normal. This behavior is not observed in the rest of the sonoff devices.

In addition, many use it in the management of garage doors, if the device switch on reset - this would lead to big problems.

joba-1, 2019-03-10:
since (most?) other sonoff devices dont exhibit this, (well, unfortunately
I have that problem with a sonoff SV) it is either hardware or your config.
Nothing that tasmota can influence.
Maybe you can sneak in a pulldown resistor (or pullup, depending on what
you need at boot), to stabilize boot behaviour.
-- 
iphone so i typo

Jason2866, 2019-03-10:
So this device is for your planned use case a bad choice. Search for Yunshan relay
This similar device to yours doesnt have this behaviour and is supported from Tasmota

meingraham, 2019-03-10:
@badigit,

I think we concluded during our Discord dialog that the device's inching configuration appears to be set using software (i.e., selecting the mode via multiple button presses). This is interfering with the operation when Tasmota is running the device. In other words, the iTead hardware reads the selected setting at boot to then set the inching mode. Tasmota doesn't have that capability by default. You would have to reverse engineer what it is that the iTead firmware does at boot to set the mode (i.e., disable inching/enable self-locking) and replicate that with a Tasmota boot time triggered rule.

Or... buy [a $6 Yunshan](https://www.aliexpress.com/item/ESP8266-Relay-Module-10A-220V-Network-Relay-WIFI-Module-Input-DC-7V-30V-4-Layers-Board/32885258262.html?spm=a2g0s.9042311.0.0.75bd4c4dOkoKgL).

Mike

Jason2866, 2019-03-10:
@meingraham  You can get it cheaper under 4€ :-) [Yunshan](https://www.aliexpress.com/item/DC-12V-ESP8266-ESP-12F-Wireless-Wifi-Module-220V-10A-Relay-Module-ESP12F-802-11b-g/32967365594.html?spm=a2g0s.13010208.99999999.259.5c9e3c001HARCW)

badigit, 2019-03-10:
@meingraham 
Hi. Im trying settings Generic device - it doesnt help. 

Yunshan requed DC 7V, Sonoff - 5V

Jason2866, 2019-03-10:
You have to change the circuit of the device. Cut the actual connected GPIO which is leading to transistor of the relais. Connect instead GPIO 5. This GPIO doesnt change his state through power cycle.
It could be you have to add a pull up resistor with 4k7 Ohm.

meingraham, 2019-03-10:
Agreed, the 7VDC is an issue. I had to buy a power adapter that could supply the needed voltage. https://express.google.com/product/5986298373924107919_5790268461285617103_12145

ascillato2, 2019-03-11:
Closing this issue as it has been answered.

Please, if you need further assistance, do not hesitate on asking here or in the Tasmota Support Chat. Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-03-11 by ascillato2