arendst/Tasmota #5430: Bug in current value (amperage)

detafun, 2019-03-09:
Hello,
I have the current version 6.4.1.20 Core 2.5.0, but no matter what core. I calibrated to Wiki, with a 75 watt incandescent lamp. 
With the 75 watt bulb is (A) ok. As soon as you plug another power source into the socket, in my example a 25 watt incandescent lamp, the (A) does not change. It remains at the value of the 75 watt incandescent lamp. See pictures attached.

[25 Watt](http://forum.creationx.de/index.php?attachment/3428-screenshot-20190309-115400-png/)

[75 Watt](http://forum.creationx.de/index.php?attachment/3429-screenshot-20190309-115316-png/)

cu Deta

localhost61, 2019-03-09:
It seems that (Active) Power is correctly reporting 25W and 75W.
Are you talking about Apparent Power? In that case it  reports effectively that your bulb is not incandescent, the reactive part is too high.
To confirm you should test more devices.

detafun, 2019-03-09:
the **Current 0.327 A**  that is the problem.
I have a dryer running on it now and the value of Current 0.327A almost does not change. Only minimal no matter how many watts are pending. Only at 0 watts, everything is at 0.

localhost61, 2019-03-09:
OK then.
For my information could you post a screenshot with the dryer?

detafun, 2019-03-09:
Yes, I can do it tomorrow. At the moment it is almost midnight and there is also the dryer at bed :-)

detafun, 2019-03-10:
There is still something wrong with the dryer. I have now parallel paralell connected an external meter. The dryer is currently active when drying the laundry and there I am at the external meter between 3-4 amps, fluctuates naturally and between 800 and 900 watts fluctuates. But the Tasmota software shows a lot less. It shows around the 0,328 A and the 76 watts.

arendst, 2019-03-10:
Go back to the beginning. 

Tasmota works fine with energy metering as long as you provide the correct input. Obviously your device configuration is either wrong or it is defect.

To continue solving your issue provide the requested information when you openend this issue. Also tell us what hardware you  use and provide the template informatipn.

localhost61, 2019-03-10:
@detafun, this is what I wanted to check, which values are wrong. It appears that both _Current_ and _Apparent Power_ are wrong. Please provide the screenshot with the dryer.

detafun, 2019-03-10:
Sorry we have to move the problem first. I had just set the devices back and the wireless PW had entered wrong now. Now I have no connection to this device. :-( I also tried it at the touch of a button> 40 seconds back does not want to set .. And you can not open the part, had flashed it via OTA. Report me again.

detafun, 2019-03-10:
No possibility to get to the device. Sorry the ticket can be made to. I'll probably have to get new others. I got the device now, but you can not get to the board because it's glued. Thus, no direct flashing possible because the ESP is not accessible. Thanks anyway.

closed 2019-03-10 by detafun

localhost61, 2019-03-10:
@detafun Sorry for you device. RIP !