arendst/Tasmota #5432: Tuya based Smart Plug with Power Monitoring

vaidyasr, 2019-03-09:
I got a Tuya based smart plug with power monitoring. I was doing trial & error on the LED & switch assignments and was able to find it sucessfully. But for power monitoring, I was trying to change the values of GPIO4, GPIO5 & GPIO12 to find the correct voltage/current values at the main page. But I can't find the correct settings to select. I was trying BL9037CF, HLWBL SEL, etc. Could someone help what which sensor is for voltage, and one for current and power.

DavinKD, 2019-03-09:
Do you have a picture or link to the plug?

vaidyasr, 2019-03-10:
The plug is here https://iswitch.online/wp-content/uploads/2019/01/16_Amp_smart_plug.jpeg

ascillato2, 2019-03-11:
Hi,

For support it is available the **Tasmota Support Chat** in which we can help you step by step and solve your questions. Github issues are for Tasmota Bug Reporting with the Template completed. Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-03-11 by ascillato2

marciogranzotto, 2020-01-02:
@vaidyasr any luck?