arendst/Tasmota #5436: Mulitple Analog Inputs

KDMcMullan, 2019-03-10:
Problem:
Among many of the TASMOTA supported devices, ESP8266 only has a single analog input. It is possible to multiplex this pin by supplying Vref to the devices being measured one at a time using GPIO pins. The outputs, then, are connected to the AIN by diodes.

Alternative:
This can be achieved by sending a sequence of MQTT commands to switch the DO pins on one at a time and reading back the AIN.

Solution:
I would like to investigate how to offload this sequencing to TASMOTA. I feel that the micro would be better placed to step through the DO pins, one at a time, sampling at the AI pin, collating the results and then publishing them as requested. Configuration would require a mechanism to instruct TASMOTA as to a list of the DO pins used.

I am a electrical and software engineer (although limited experience in C) and would be interested in assisting with the development of any solution.

Thanks for a fantastic piece of work. I have TASMOTA on Sonoff T1, TH10 (stock and modified), 433MHz bridge, Slampher and am about to drop it on a NodeMCU.

Jason2866, 2019-03-10:
With already supported ADS1115 you get 4 analog inputs via I2C

ascillato2, 2019-03-10:
Thanks for sharing your ideas.

Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-03-10 by ascillato2