arendst/Tasmota #5437: `setoption19 1` restartes Tasmato, makes decive unavailable in Home Assistant

rklasen, 2019-03-10:
<GUIDE>

<This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum>
  
<Please DO NOT OPEN AN ISSUE:>
  <If you have general questions or you need help on Tasmota usage, go to the Tasmota support chat>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a new device, please use the Tasmota Template Feature. See wiki for that>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### ISSUE DESCRIPTION - TROUBLESHOOTING

I flashed a gosund SP1 (BlitzWolf SP1 clone, ESP8285) with the latest Tasmato firmware 6.4.1 using esptool.py. I added the MQTT configuration and in Home Assistant, I can see the device connects. When I enter `setoption19 1` into Tasmatos console, the gosund device restarts (I don't know if it's supposed to), and it shows up in Home Assistant as a switch. However, it's status is unavailable and switing doesn't work.

When I manually add the device to `configuration.yaml`:

```
switch:
  - platform: mqtt
    name: "Schlafzimmer LEDs Manuell"
    state_topic: "stat/sz/schalter/licht/POWER"
    command_topic: "cmnd/sz/schalter/licht/POWER"
    availability_topic: "tele/sz/schalter/licht/LWT"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    payload_available: "Online"
    payload_not_available: "Offline"
    retain: true
```

it works, but I'm missing the power monitor functions.

Is the gosund device supposed to restart after I `setoption19 1`? Do I have to change some configuration in Home Assistant to make it available?

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Gosund SP1 v23
- [x] Tasmota binary firmware version number used : 6.4.1 / (pre-compiled)
- [x] Development IDE - Compiler / Upload tools used : none / esptool.py
- [x] Provide the output of command ``status 0`` :
```
18:01:54 CMD: status 0
18:01:54 MQT: stat/sz/schalter/licht/STATUS = {"Status":{"Module":55,"FriendlyName":["sw.schlafz.licht"],"Topic":"/sz/schalter/licht","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:01:54 MQT: stat/sz/schalter/licht/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-DE.bin","RestartReason":"Software Watchdog","Uptime":"0T00:01:33","StartupUTC":"2019-03-10T17:00:21","Sleep":50,"BootCount":39,"SaveCount":135,"SaveAddress":"F8000"}}
18:01:54 MQT: stat/sz/schalter/licht/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018.12.24 14:55:04","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
18:01:54 MQT: stat/sz/schalter/licht/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Gast",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
18:01:54 MQT: stat/sz/schalter/licht/STATUS4 = {"StatusMEM":{"ProgramSize":522,"Free":480,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000407","0FDAE794","000383A0","23B617CE","00003BC0"]}}
18:01:54 MQT: stat/sz/schalter/licht/STATUS5 = {"StatusNET":{"Hostname":"/sz/schalter/licht-2800","IPAddress":"192.168.1.230","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.2","Mac":"84:0D:8E:53:CA:F0","Webserver":2,"WifiConfig":4}}
18:01:54 MQT: stat/sz/schalter/licht/STATUS6 = {"StatusMQT":{"MqttHost":"hassio.lan","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_53CAF0","MqttUser":"tasmota","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
18:01:54 MQT: stat/sz/schalter/licht/STATUS7 = {"StatusTIM":{"UTC":"Sun Mar 10 17:01:54 2019","Local":"Sun Mar 10 18:01:54 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:14","Sunset":"18:46"}}
18:01:54 MQT: stat/sz/schalter/licht/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
18:01:55 MQT: stat/sz/schalter/licht/STATUS10 = {"StatusSNS":{"Time":"2019-03-10T18:01:54","ENERGY":{"TotalStartTime":"2019-03-09T10:37:54","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
18:01:55 MQT: stat/sz/schalter/licht/STATUS11 = {"StatusSTS":{"Time":"2019-03-10T18:01:55","Uptime":"0T00:01:34","Vcc":3.399,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Gast","BSSId":"12:FE:ED:93:16:7C","Channel":11,"RSSI":42}}}
```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```
**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-03-10:
Hi,

This was already solved in latest version of the development branch.

Please update to latest TASMOTA from: http://thehackbox.org/tasmota/

Thanks.

rklasen, 2019-03-10:
Hi, thanks for your fast reply. I just updated to the latest version, but the issue persists. From what I can see from the `status 0` output:

```
20:17:43 CMD: status 0
20:17:43 MQT: stat/wz/schalter/licht1/STATUS = {"Status":{"Module":55,"FriendlyName":["sw.wohnz.licht"],"Topic":"/wz/schalter/licht1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:17:43 MQT: stat/wz/schalter/licht1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software Watchdog","Uptime":"0T00:00:11","StartupUTC":"2019-03-10T19:17:32","Sleep":50,"BootCount":28,"SaveCount":122,"SaveAddress":"F7000"}}
20:17:43 MQT: stat/wz/schalter/licht1/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T14:41:00","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
20:17:43 MQT: stat/wz/schalter/licht1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Gast",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
20:17:43 MQT: stat/wz/schalter/licht1/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
20:17:43 MQT: stat/wz/schalter/licht1/STATUS5 = {"StatusNET":{"Hostname":"/wz/schalter/licht1-1254","IPAddress":"192.168.1.221","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.2","Mac":"84:0D:8E:53:64:E6","Webserver":2,"WifiConfig":4}}
20:17:43 MQT: stat/wz/schalter/licht1/STATUS6 = {"StatusMQT":{"MqttHost":"hassio.lan","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_5364E6","MqttUser":"tasmota","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:17:43 MQT: stat/wz/schalter/licht1/STATUS7 = {"StatusTIM":{"UTC":"Sun Mar 10 19:17:43 2019","Local":"Sun Mar 10 20:17:43 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:14","Sunset":"18:46"}}
20:17:43 MQT: stat/wz/schalter/licht1/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
20:17:43 MQT: stat/wz/schalter/licht1/STATUS10 = {"StatusSNS":{"Time":"2019-03-10T20:17:43","ENERGY":{"TotalStartTime":"2019-03-09T10:37:54","Total":0.055,"Yesterday":0.000,"Today":0.055,"Power":43,"ApparentPower":76,"ReactivePower":63,"Factor":0.57,"Voltage":234,"Current":0.326}}}
20:17:43 MQT: stat/wz/schalter/licht1/STATUS11 = {"StatusSTS":{"Time":"2019-03-10T20:17:43","Uptime":"0T00:00:11","Vcc":3.462,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"Gast","BSSId":"12:FE:ED:93:16:7C","Channel":11,"RSSI":78}}}
```

this firmware is still compiled against commit cfd48f3, although the web site states d0ac200. Maybe I'm understanding this wrong though. Thanks!

rklasen, 2019-03-10:
Oh, I just realized that ota update didn't work! I updated to the minimal image and from there to the current image, and the issue is resolved indeed. Thanks again!

closed 2019-03-10 by rklasen