arendst/Tasmota #5438: Tasmota 6.4.1, Wemos d1 MINI, knx, multiple DS18B20

MachmaHinne, 2019-03-10:
Hi,

as shown in the subject, I´m using a Wemos D1 mini with Tasmota 6.4.1 and two DS18B20 temperature sensors. I´m also using knx as the communication protocol. Mapping one temperature sensor to an group adress was no problem.
How can I map the second sensor to another group adress?
If not possible, please add this as a new feature
Thank you
Greetings

[image]
[image]
[image]
[image]

ascillato, 2019-03-11:
Hi

Yes, it is possible using rules. 

See the example of the wiki at https://github.com/arendst/Sonoff-Tasmota/wiki/KNX-Features#6-rule-to-send-knx-telegram-with-bh1750-sensor-data and adapt it to your sensor.

It is not going to be added to the knx menu due to flash and RAM constraints.

Thanks. 

If you need further assistant, please find us in the wiki.

ascillato2, 2019-03-11:
### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-03-11 by ascillato2

MachmaHinne, 2019-03-11:
Hi Adrian,

Thank you for your help.
I´m trying it with:

rule1 on  DS18B20-2#Temperature do knxtx_val1 %value% endon

but the Temperature is not send to the GAD that is connected to knx_tx1.
I manually send a value to knx_tx1 to verify that the GAD is working, and it is.
Status says, that the sensor is there:

23:57:27 RSL: SENSOR = {"Time":"2019-03-11T23:57:27","DS18B20-1":{"Id":"0416807B2AFF","Temperature":22.6},"DS18B20-2":{"Id":"0516805884FF","Temperature":22.7},"TempUnit":"C"}

Greetings
Peter

ascillato, 2019-03-12:
Hi,

You can use the teleperiod trigger to send data like:

```
rule1 1
rule1 on tele-DS18B20-2#Temperature do knxtx_val1 %value% endon
```

Another option is to adapt the example of the wiki that send temperature values on a delta.

Please, tell us if this solves your issue and if you need further assistant. In that case we can reopen this issue.

MachmaHinne, 2019-03-12:
Yes, it works! I´m very happy with it.
Thank you very much.

Greetings

Peter

MachmaHinne, 2019-03-13:
Log:
working for 1 day without failure.

MachmaHinne, 2020-06-07:
In addition, if multiple DS18B20 have to send values:

`rule1 1
rule1 on tele-DS18B20-1#Temperature do knxtx_val1 %value% endon on tele-DS18B20-2#Temperature do knxtx_val2 %value% endon on tele-DS18B20-3#Temperature do knxtx_val3 %value% endon`