arendst/Tasmota #5441: Exception if Baudrate set to 300

Jason2866, 2019-03-11:
### BUG DESCRIPTION
Setting Sbaudrate to 300 causes exception

### REQUESTED INFORMATION
Type `Sbaudrate 300` in console

`tele/sonoff-53CF5F/INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x4000dce5 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}`

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x ] Device used (i.e. Sonoff Basic) : wemos mini
- [x ] Tasmota binary firmware version number used : ____ / (pre-compiled or self-compiled ?)
- [ x] Development IDE - Compiler / Upload tools used : ____ / ____
- [ x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

```
19:45:51 CMD: Status 0
19:45:51 MQT: stat/sonoff-53CF5F/STATUS = {"Status":{"Module":18,"FriendlyName":["169"],"Topic":"sonoff-53CF5F","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:45:51 MQT: stat/sonoff-53CF5F/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.2.20/data/firmwares/sonoff.bin","RestartReason":"Exception","Uptime":"0T00:03:06","StartupUTC":"2019-03-11T18:42:45","Sleep":50,"CfgHolder":4617,"BootCount":23,"SaveCount":58,"SaveAddress":"F6000"}}
19:45:51 MQT: stat/sonoff-53CF5F/STATUS2 = {"StatusFWR":{"Version":"6.4.1.21(sonoff)","BuildDateTime":"2019-03-11T17:49:59","Boot":31,"Core":"STAGE","SDK":"2.2.1(cfd48f3)"}}
19:45:51 MQT: stat/sonoff-53CF5F/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Jason_Home_WLAN",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
19:45:51 MQT: stat/sonoff-53CF5F/STATUS4 = {"StatusMEM":{"ProgramSize":494,"Free":508,"Heap":26,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","0F8AA784","040403A1","80041496","001028C2"]}}
19:45:51 MQT: stat/sonoff-53CF5F/STATUS5 = {"StatusNET":{"Hostname":"sonoff-53CF5F-3935","IPAddress":"192.168.2.169","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"B4:E6:2D:53:CF:5F","Webserver":2,"WifiConfig":5}}
19:45:51 MQT: stat/sonoff-53CF5F/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.114","MqttPort":1883,"MqttClientMask":"sonoff-53CF5F","MqttClient":"sonoff-53CF5F","MqttUser":"openhabian","MqttType":1,"MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:45:51 MQT: stat/sonoff-53CF5F/STATUS7 = {"StatusTIM":{"UTC":"Mon Mar 11 18:45:51 2019","Local":"Mon Mar 11 19:45:51 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"06:35","Sunset":"18:11"}}
19:45:51 MQT: stat/sonoff-53CF5F/STATUS10 = {"StatusSNS":{"Time":"2019-03-11T19:45:51"}}
19:45:51 MQT: stat/sonoff-53CF5F/STATUS11 = {"StatusSTS":{"Time":"2019-03-11T19:45:51","Uptime":"0T00:03:06","Vcc":2.945,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Dimmer":99,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:2A:BD:19","Channel":13,"RSSI":94,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```

arendst, 2019-03-12:
Baudrates below 1200 are not supported. I acknowledge the test for lower baudrates is wrong and will fix the exception.

Do you really need baudrates below 1200 bps?

Saimen74, 2019-03-12:
Hi arendst, 
thanks for fixing the exception. 
I want to use this [IR-Reading Header](https://www.ebay.de/itm/TTL-IR-Infrarot-Lese-Schreibkopf-Stromzahler-Smart-Meter-/273582906569)  connected to an Wemos D1  for reading my power meter [Pafal 20EC3gr](https://www.symcon.de/forum/threads/31166-Stromz%C3%A4hler-Pafal-20EC3gr-auslesen)

The Pafal uses 300 baud 300,7,1,Even to communicate.

It would be really nice to use this great Tasmota Firmware as a Serial Bridge. 

Michael

arendst, 2019-03-12:
Tasmota supports baudrates downto 1200 bps due to design decisions I made.

Your IR-Reading Header supports up to 19200 so that shouldn't be a problem.

ascillato2, 2019-03-12:
Closing this issue as it has been fixed. Thanks

closed 2019-03-12 by ascillato2