arendst/Tasmota #5442: Setting rule with backlog not working

herrfrei, 2019-03-11:
### BUG DESCRIPTION
I'm trying to set up a rule for a MP3 speaker with backlog to separate two commands for one event. Everything after the first command is ignored.

Rule1 has the following contents:
```
on System#Boot do 
  backlog MP3Volume 75;MP3Track 1 
endon 
on Wifi#Connected do 
  MP3Track 2 
endon 
on Wifi#Disconnected do 
  MP3Track 3 
endon 
on Mqtt#Connected do 
  MP3Track 4 
endon 
on Mqtt#Disconnected do 
  MP3Track 5 
endon
```
Sending this via cmnd/.../backlog results in the following messages recording with mosquitto_sub:

```
cmnd/speaker/backlog Rule1 on System#Boot do backlog MP3Volume 75; MP3Track 1 endon on Wifi#Connected do MP3Track 2 endon on Wifi#Disconnected do MP3Track 3 endon on Mqtt#Connected do MP3Track 4 endon on Mqtt#Disconnected do MP3Track 5 endon; Rule1 4; Rule1 8; Rule1 1; 
stat/speaker/RESULT {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":473,"Rules":"on System#Boot do backlog MP3Volume 75"}
stat/speaker/RESULT {"MP3Track":1}
stat/speaker/RESULT {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":473,"Rules":"on System#Boot do backlog MP3Volume 75"}
stat/speaker/RESULT {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":473,"Rules":"on System#Boot do backlog MP3Volume 75"}
stat/speaker/RESULT {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":473,"Rules":"on System#Boot do backlog MP3Volume 75"}
```
### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Wemos D1 Mini
- [x] Tasmota binary firmware version number used : 6.4.1 / pre-compiled for sensors
- [x] Development IDE - Compiler / Upload tools used: ESP.Easy.Flasher.exe
- [x] Provide the output of command ``status 0`` :
```
20:14:59 CMD: status 0
20:14:59 MQT: stat/speaker/STATUS = {"Status":{"Module":18,"FriendlyName":["MP3Speaker"],"Topic":"speaker","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:14:59 MQT: stat/speaker/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-sensors.bin","RestartReason":"External System","Uptime":"0T00:14:58","StartupUTC":"2019-03-11T19:00:01","Sleep":50,"BootCount":44,"SaveCount":73,"SaveAddress":"FB000"}}
20:14:59 MQT: stat/speaker/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sensors)","BuildDateTime":"2018-12-24T14:45:55","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
20:14:59 MQT: stat/speaker/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Radio Jerewan",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
20:15:00 MQT: stat/speaker/STATUS4 = {"StatusMEM":{"ProgramSize":550,"Free":452,"Heap":15,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","0FDEE794","0003A3A4","B7FFBFCC","0002BBC0"]}}
20:15:00 MQT: stat/speaker/STATUS5 = {"StatusNET":{"Hostname":"speaker-2718","IPAddress":"192.168.178.86","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"CC:50:E3:2B:4A:9E","Webserver":2,"WifiConfig":4}}
20:15:00 MQT: stat/speaker/STATUS6 = {"StatusMQT":{"MqttHost":"homesrv","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2B4A9E","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:15:00 MQT: stat/speaker/STATUS7 = {"StatusTIM":{"UTC":"Mon Mar 11 19:15:00 2019","Local":"Mon Mar 11 20:15:00 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:12","Sunset":"18:48"}}
20:15:00 MQT: stat/speaker/STATUS10 = {"StatusSNS":{"Time":"2019-03-11T20:15:00","ANALOG":{"A0":2}}}
20:15:00 MQT: stat/speaker/STATUS11 = {"StatusSTS":{"Time":"2019-03-11T20:15:00","Uptime":"0T00:14:59","SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"Wifi":{"AP":1,"SSId":"________","BSSId":"xx:4E:xx:59:xx:65","Channel":11,"RSSI":64}}}
```
- [x] Provide the output of console when you experience your issue if apply :
Setting weblog to 4 gives the following console output:

```
20:15:48 CMD: weblog 4
20:15:48 MQT: stat/speaker/RESULT = {"WebLog":4}
20:15:48 CFG: Saved to flash at FA, Count 74, Bytes 3584
20:15:56 WIF: Checking connection...
20:15:56 WIF: Connected
20:15:57 SRC: MQTT
20:15:57 RSL: Received Topic cmnd/speaker/backlog, Data Size 248, Data Rule1 on System#Boot do backlog MP3Volume 75;MP3Track 1 endon on Wifi#Connected do MP3Track 2 endon on Wifi#Disconnected do MP3Track 3 endon on Mqtt#Connected do MP3Track 4 endon on Mqtt#Disconnected do MP3Track 5 endon; Rule1 4; Rule1 8; Rule1 1; 
20:15:57 RSL: Group 0, Index 1, Command BACKLOG, Data Rule1 on System#Boot do backlog MP3Volume 75;MP3Track 1 endon on Wifi#Connected do MP3Track 2 endon on Wifi#Disconnected do MP3Track 3 endon on Mqtt#Connected do MP3Track 4 endon on Mqtt#Disconnected do MP3Track 5 endon; Rule1 4; Rule1 8; Rule1 1; 
20:15:57 SRC: Backlog
20:15:57 RSL: Received Topic /Rule1, Data Size 38, Data on System#Boot do backlog MP3Volume 75
20:15:57 RSL: Group 0, Index 1, Command RULE, Data on System#Boot do backlog MP3Volume 75
20:15:57 MQT: stat/speaker/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":473,"Rules":"on System#Boot do backlog MP3Volume 75"}
20:15:57 SRC: Backlog
20:15:57 RSL: Received Topic /MP3Track, Data Size 165, Data 1 endon on Wifi#Connected do MP3Track 2 endon on Wifi#Disconnected do MP3Track 3 endon on Mqtt#Connected do MP3Track 4 endon on Mqtt#Disconnected do MP3Track 5 endon
20:15:57 RSL: Group 0, Index 1, Command MP3TRACK, Data 1 endon on Wifi#Connected do MP3Track 2 endon on Wifi#Disconnected do MP3Track 3 endon on Mqtt#Connected do MP3Track 4 endon on Mqtt#Disconnected do MP3Track 5 endon
20:15:58 MQT: stat/speaker/RESULT = {"MP3Track":1}
20:15:58 SRC: Backlog
20:15:58 RSL: Received Topic /Rule1, Data Size 1, Data 4
20:15:58 RSL: Group 0, Index 1, Command RULE, Data 4
20:15:58 MQT: stat/speaker/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":473,"Rules":"on System#Boot do backlog MP3Volume 75"}
20:15:58 SRC: Backlog
20:15:58 RSL: Received Topic /Rule1, Data Size 1, Data 8
20:15:58 RSL: Group 0, Index 1, Command RULE, Data 8
20:15:58 MQT: stat/speaker/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":473,"Rules":"on System#Boot do backlog MP3Volume 75"}
20:15:59 SRC: Backlog
20:15:59 RSL: Received Topic /Rule1, Data Size 1, Data 1
20:15:59 RSL: Group 0, Index 1, Command RULE, Data 1
20:15:59 MQT: stat/speaker/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":473,"Rules":"on System#Boot do backlog MP3Volume 75"}
20:16:17 WIF: Checking connection...
20:16:17 WIF: Connected
```

### TO REPRODUCE
- Send a rule setup command as described at the beginning of the issue report.

### EXPECTED BEHAVIOUR
Rules should be returned and executed as entered

ascillato, 2019-03-11:
Sorry, nested backlogs are not supported as explained in the wiki.

ascillato, 2019-03-11:
You will need to send those commands separatedly

herrfrei, 2019-03-11:
Found a rule like this in the wiki. What's the difference to my rule?

Rule on tele-SI7021#temperature do backlog var1 %value%; add1 2; event sendtemp endon on event#sendtemp do publish stat/sonoff

herrfrei, 2019-03-11:
May be it's an issue resulting from usage the Tasmota Device Manager (tdm). It sends a rule plus its settings (on, once,... ) as a backlog.?

herrfrei, 2019-03-11:
It's working when rules are sent without cmnd/.../backlog. I'll open this as issue on tdm now.

closed 2019-03-11 by herrfrei

jeroenst, 2021-10-11:
> It's working when rules are sent without cmnd/.../backlog. I'll open this as issue on tdm now.

How did you achieve this? I have the same problem although backlog is stated in the manual.

```
Backlog SwitchMode 11; SetOption32 10; Rule1 1;

Rule1 
on system#boot do var1 + ENDON
on switch1#state=2 do backlog DIMMER 80; POWER TOGGLE ENDON
on switch1#state=4 do DIMMER %var1% ENDON
on switch1#state=7 do event upordown=%var1% ENDON
on event#upordown=+ do var1 - ENDON
on event#upordown=- do var1 + ENDON
```