arendst/Tasmota #5443: Support for Sonoff S26F (not S26)

MatzeMuc86, 2019-03-11:
I just realized Sonoff offers S26F instead of S26 devices as well. They can be flashed successfully similar to the S26 but the relay will not work, even though one can hear its action.

Therefore I would like to ask for supporting the Sonoff S26F device as well (IM180320004)

Thank you and best
MatzeMuc86

ascillato, 2019-03-11:
>the relay will not work, even though one can hear its action

That is weird. If you can hear the click of the relay, it is working. Did you tested that device before flashing Tasmota?

MatzeMuc86, 2019-03-11:
> > the relay will not work, even though one can hear its action
> 
> That is weird. If you can hear the click of the relay, it is working. Did you tested that device before flashing Tasmota?

Yes I tested the original firmware before and the relay (as well as the device) were working as expected. I have three of the devices and flashed two of them already. These two devices behave the same (hearing relay but no action)

ascillato, 2019-03-11:
Do you have pictures of that device? I couldn't find it in the official iTead page.

KPK75, 2019-03-11:
It's just a name on Aliexpress used for socket type ( S26E, S26F etc. ). Mine S26E works fine.

MatzeMuc86, 2019-03-11:
Attached one picture of the device as well as one picture of the box.
[image]
[image]

MatzeMuc86, 2019-03-11:
> It's just a name on Aliexpress used for socket type ( S26E, S26F etc. ). Mine S26E works fine.

Could be the case. I just realized the blinking LED might be not regular: https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#relay-clicksled-flashes-at-1-sec-interval

Sorry if I caused confusion by the F of the S26F (but at least people can not find answers via searching/google).

Any hint for the LED blinking (which sounds more realistic to indicate the issue). I am quite confused as the firmware flashing seams to be successful, web server works, settings persist after reboot etc.

Thank you and best

arendst, 2019-03-11:
How do you test it functionality? Using it as designed as a socket connected to AC or only via the USB serial connection.

MatzeMuc86, 2019-03-11:
> How do you test it functionality? Using it as designed as a socket connected to AC or only via the USB serial connection.

The original FW was working for several months. I first tested the Tasmota FW via USB and felt successful as I heard the relay. Then I tried via AC and can still hear the relay but nothing is working.

I just had the idea not upload the original firmware to ensure I have a proper SW without any local buts but it states there is not enough free memory.

Do you have any ideas / hints?

Thank you

MatzeMuc86, 2019-03-11:
> I just had the idea not upload the original firmware to ensure I have a proper SW without any local buts but it states there is not enough free memory.

I actually tried the sonoff.bin, sonoff-classic.bin as well as the sonoff-minimal.bin. All of them with the same error message.

MatzeMuc86, 2019-03-11:
I tried the other one of the two devices. The interesting fact is that the LED is NOT blinking (maybe that's why I have not realized the other one blinking before).
I have generated a video that shows you the behavior. At the end I manually (and carefully!) disconnect the AC light and you can see it is getting darker.

Check out the video here: https://www.dropbox.com/s/gb0wux8zbqolrnn/20190311_223721.mp4?dl=0

I am quite confused what the real problem might be.

MatzeMuc86, 2019-03-11:
> > I just had the idea not upload the original firmware to ensure I have a proper SW without any local buts but it states there is not enough free memory.
> 
> I actually tried the sonoff.bin, sonoff-classic.bin as well as the sonoff-minimal.bin. All of them with the same error of not enough memory.

Same actually for the second (not blinking) device for all three bin files.

So I am quite at the end of ideas. Any suggestions?

Thank you and best
MatzeMuc86

MatzeMuc86, 2019-03-11:
Last thought from my side for today: is there anyhow any chance the fw can be wrong if is is flashed and the web server is running? I just try to sort out potential root causes.

joba-1, 2019-03-12:
2 ideas:

With not working, you just mean devices plugged in get no power? Al least you can somehow operate the relay, right? Then I guess the soldered fuse is blown.

The not enough mem during reflash could be caused by not enough low volt power via the flash adapter. You could try a better, separate 3.3V power supply and just connect tx/rx/gnd of your flash adapter if you want to reflash.

blakadder, 2019-03-12:
You should try to erase flash with esptool and reflash a precompiled latest dev binary. If that doesn't solve the issue you will have to trace the pins and see if the pinout is the same as in the wiki.

ascillato2, 2019-03-14:
Hi, any news on this?

MatzeMuc86, 2019-03-14:
Sorry for the late reply.

I investigated a lot the last few days. Actually I am quite sure the fw is flashed correctly as I can access and operate the relay via the web interface. 
My questions and errors regarding flashing were via OTA. Maybe some hints on that? 

For the original topic of the non functional but hearable relay it seems to be a pure (stupid) physical problem: I tried the device disassembled and the connector of the device I want to connect can get direct contact to the input ac metal plate. I'll send some pictures later this week as it is simple but hard to put into words. So anybody who has similar issues: assemble the device properly for correct testing, not like me! 

The blinking led seems to be rather a problem of mqtt of one of the devices, the second one does not blink. 

I would be very thankful for hints regarding too less memory for Ota flashing. 

Thank you and best

ascillato2, 2019-03-14:
Closing as this issue was solved.

About OTA upgrade, if you use otaurl it is automatic, if you use ota by file you need to use minimal first and then the full version. More information please check the wiki. If you need further assistant, please find us in the Tasmota support chat. Github issues are for Tasmota Software Bug reporting with the template completed. Thanks

closed 2019-03-14 by ascillato2