arendst/Tasmota #5444: Tuya QS-WIFI-S03 / Shelly clone

tifrei, 2019-03-11:
Hello,

bought this shelly clone for testing. Already got tasmota uploaded but im not sure on how to set this up so that it works properly ?

Any ideas from you Tasmota God's?

<blockquote class="imgur-embed-pub" lang="en" data-id="a/QHZDGfl"><a href="//imgur.com/QHZDGfl"></a></blockquote><script async src="//s.imgur.com/min/embed.js" charset="utf-8"></script>

https://imgur.com/BwFbq4w

https://imgur.com/s13yEXT

netpok, 2019-03-12:
You have to trace the switch contact and relay driver transistor back to a gpio and set them up as switch/relay in module settings.

blakadder, 2019-03-12:
Please try and ask for help in [Discord chat](https://discord.gg/Ks2Kzd4), issues are meant for bugs in the code not individual help

closed 2019-03-12 by blakadder