arendst/Tasmota #5445: Ph sensor 4502C readout

superjasiek, 2019-03-11:
Hi! thanks for your grate work - it's fantastic and makes relay easy to integrate stuff with home assistant. Up till now i'm using tasmota to monitor my aquarium - temp readings  and shrimp feeder. I bought also ph sensor but can't get it working with tasmota. It uses analog pin to get readings from ph electrode but they need to be converted. And here is a question from newbie - is there any way to integrate the code below with tasmota?

```
const int pinPo = A0;

void setup() {
  // serial communication 9600 baud
  Serial.begin(9600);
}

void loop() {
  // variables 
  int pole[10];
  int zaloha;
  unsigned long int prumerVysl = 0;
// load ten samples for 10 ms into the array
  for (int i = 0; i < 10; i++) {
    pole[i] = analogRead(pinPo);
    delay(10);
  }
  // sort the members of the field of measured results by size
  for (int i = 0; i < 9; i++) {
    for (int j = i + 1; j < 10; j++) {
      if (pole[i] > pole[j]) {
        zaloha = pole[i];
        pole[i] = pole[j];
        pole[j] = zaloha;
      }
    }
  }
  // saving 2nd to 8th result to
   // the variable from which the average is calculated
   // (omit two field members at the beginning
   // and end for better accuracy)
  for (int i = 2; i < 8; i++) {
    prumerVysl += pole[i];
  }
 // calculate the pH value from the average
   // Measurement and conversion to 0-14 pH range
  float prumerPH = (float)prumerVysl * 5.0 / 1024 / 6;
  float vyslednePH = -5.70 * prumerPH + 21.34;
 // print out the results on the serial line
  Serial.print("Pomiar pH: ");
  Serial.println(vyslednePH);
 // pause 900 ms before new measurement, total 1s
  delay(900);
}
```
This code works well with arduino uno but it would more useful to use it with wireless controller.
Take care!

Frogmore42, 2019-03-12:
The simple answer is that it would not be easy to do this.

You did not provide any information about what the sensor actually outputs.  A normal Arduino has a 5V maximum input on its ADC.  The esp8266 has a maximum of 1V, but the NodeMCU variant has a resistor divider which provides a 3V3 maximum.

The other thing is the standard ADC reading on Tasmota does not do any averaging, let alone discarding the top and bottom values.

Now, it is possible to write a special sensor routine that does averaging and the conversion to pH.  The conversion to pH could possibly be done as a rule and the averaging could be done in your Home Automation system, but it would be different than the code you provided.

superjasiek, 2019-03-12:
That's true - 

> The esp8266 has a maximum of 1V, but the NodeMCU variant has a resistor divider which provides a 3V3 maximum.

In general output from pH sensor is voltage where 0-5V represents 0-14 pH so more less 2,5V is pH 7
I think it is also possible to use external power supply and use logic converter(?)
Any way right now i'm testing reading the pH value from arduino with nodeMCU  and Tasmota - seems to work fine but aquarium monitoring system already have 3 boards ;-)

Jason2866, 2019-03-15:
Code can not be implemented because voltage is different and so the formular is wrong for ESP.
Please close this issue.
Thx

superjasiek, 2019-03-15:
So no other solution for Tasmota except reading serial port from arduino? Closing than.

closed 2019-03-15 by superjasiek

sunflowerABB, 2019-09-07:
I would also be interested in being able to read the "PH-4502c" sensor using Tasmota. Pinout & schematics can be found easily online...

godysseus, 2020-01-05:
Someone had success with the PH 4502C ? Tasmota integration would be really really great.

Lumoc, 2020-04-03:
Use it via a ADC pin and calculate the value by a rule? To calibrate it use a 200K OHM Linear Taper Poti on a Wemos D1 mini to reduce the 5V to 3.3V max (around 180k)
Measured voltage -> Actual pH (buffer solution)
          - 0.59 -> 7.0
          - 0.71 -> 4.0

Hope this helps, will try myself as soon as the parts arrive.

effelle, 2020-04-03:
To convert data from A0 you can also select `ADCRange` from the drop down menu and use `AdcParam6` to scale the value to suit your needs.