arendst/Tasmota #5446: Sonoff Basic SwitchMode Not working.. on Sonoff Basic RF2 Version

asking23, 2019-03-12:
Hi,
I have flashed my Sonoff basic with Tasmota 6.4.1 successfully and its working.
I want to Add Switch to GPIO14 (Soldered wire to KEY Point on PCB) .

I have tried all the SwitchMode0....4 all commands accepted in webui console but not effective.

When i short GPIO14 and GND it toggles only. No SwitchMode is effective. what can i do ?

Frogmore42, 2019-03-12:
Issues are for real issues with the software, not for help on how to use the software.  The template provides several resources to check to get help with how to use and configure things.

ascillato2, 2019-03-12:
Closing issue as it does not follow the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md)

closed 2019-03-12 by ascillato2