arendst/Tasmota #5448: Switch1 not always working

Lwyrn, 2019-03-12:
Hello,
I've flashed Tasmota 6.4.1 to a Sonoff Basic R2 succesfully.
I'm using the GPIO1 and GPIO3 as I2C pins for a BME280 temperature sensor.
For the switch I've desoldered the status LED to get the GPIO13 as usable IO ( due to missing GPIO14 on R2 ) for the wall switch button. Setting the Sonoff as Generic (18) and change the GPIO13 to Switch1. It's working but sometimes it miss the change of the switch. I can press the button of the switch several times continuously and it works great but, sometime, i press the button and nothing happen. Then I press again the button two or tree times and it return to work again. Seems to be a random issue. The MQTT Server is correctly configured and working great. There is no errors in the log ( i've enabled debug infos ).

What can be wrong?
Thanks

netpok, 2019-03-12:
Is this pin directly connected to the ESP without current limiting resistor or drive transistor, also where do you connect the other end of the switch? Sounds more like a hardware problem for me, you should measure the open voltage on the pin, it should be around 3.3V.

Lwyrn, 2019-03-12:
IT is connected to the GPIO13 pin using a 1k resistor (R16) ( measured using the multimeter ). Should not be a problem.  Or am I wrong?

netpok, 2019-03-12:
It probably shouldn't (as the internal pullup is around ~33k if I know correctly), but just in case try to measure the esp end of the resistor. The open voltage should be 3.3 volt the closed should be close to 0v (let's say less than 0.5V)

Lwyrn, 2019-03-12:
The open oscillates to 3.1x and 3.260 ( I assume it's 3.3, it's a cheap multimeter ). The Closed is 0

netpok, 2019-03-12:
Then it should work (the voltage is okay, it can be the power supply too), what happens if you connect the switch to RX or TX?

Lwyrn, 2019-03-12:
I've tried to set "Sleep" to 0 ( using sleep 0 in the console ) and works great. Than i've tried to reset the sleep to 50 and now works great at 50ms too. I can turn OFF and ON 100 times per seconds and it don't miss a click. Tested on 2 Sonoff. But is a strange behaviour. Apparently nothing has changed, I've only changed the sleep but restored to 50. I want to do another test after a while to check if the issue come back. But I think there is a sort of software issue

ascillato, 2019-03-12:
>I can turn OFF and ON 100 times per seconds and it don't miss a click.

Your relay and Tasmota are not meant for that usage. You will burn your relay. 

When you said that the push of the button is missed, you mean that the relay were not switching or that the message by mqtt was not sent? To know that, you need to set weblog to 4 and see the messages in the console. 

If that was not working before and now it is working, seems to be just a soldering issue.

Lwyrn, 2019-03-12:
> Your relay and Tasmota are not meant for that usage. You will burn your relay.
It's just to say that it works great, i'm not intended to use it this way :D
I've noticed the issue because i've manually turned on ( using wall switch ) the light and it not turned ON.

When i say that the push of button is missed, i mean that the relay not change the state and no action get recorded to the log ( nothing happen ).

I've setted the weblog to 4 and when the action is missed, no log is reported.
I've tried to sold a pinhead to the Sonoff and using a Dupont cable the result is the same.

I've one Sonoff with 50ms of sleep and one with 0ms now. The one using 0ms is working fine after a while. On the 50ms one the issue is coming back. I've setted the sleep to 0 and now works fine.
I'll do the test again in 10-15 minutes

joba-1, 2019-03-12:
would be good to switch roles: other sonoff now with sleep 50. This could
(more or less) rule out hardware issues.
-- 
iphone so i typo

Lwyrn, 2019-03-12:
I've tested several times and seems it works fine with sleep disabled. I'm doing more tests

ascillato, 2019-03-12:
Do you have set dynamic sleep or normal sleep?
If you don't know, please share the output of status 0 command as the troubleshooting template states.

Lwyrn, 2019-03-12:
Is Dynamic mode

ascillato, 2019-03-12:
I saw the behaviour you reported just with normal sleep at 50. With dynamic I can't reproduce this issue. Anyway, please, try sleep 5. That amount is still saving a lot of energy and the response should be super fast. Tell us if that works for you. Thanks

Lwyrn, 2019-03-12:
I'm trying using sleep at 5. It seems to work fine as well. I'll do more test

ascillato2, 2019-03-12:
Great to know that now it is working for you.

Closing issue as it is solved. 

If you need further assistant, please do not hesitate on asking here or in the Tasmota support chat. Thanks

closed 2019-03-12 by ascillato2