arendst/Tasmota #5449: Sonoff T1 3CH US Ghost Switching

Bloodyagent, 2019-03-12:
- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Sonoff T1 3CH_
- [x] Tasmota binary firmware version number used : 6.4.1-2.3.0 pre-compiled bin
- [x] Development IDE - Compiler / Upload tools used : esptool
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
18:56:44 MQT: stat/TestBench/STATUS = {"Status":{"Module":30,"FriendlyName":["Sonoff","Sonoff2","Sonoff3"],"Topic":"TestBench","ButtonTopic":"0","Power":2,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:56:44 MQT: stat/TestBench/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T02:00:38","StartupUTC":"2019-03-12T15:56:06","Sleep":50,"BootCount":4,"SaveCount":92,"SaveAddress":"F8000"}}
18:56:44 MQT: stat/TestBench/STATUS2 = {"StatusFWR":{"Version":"6.4.1(release-sonoff)","BuildDateTime":"2018-12-24T15:51:34","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
18:56:44 MQT: stat/TestBench/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Byterite",""],"TelePeriod":300,"SetOption":["0000A009","558180C0","00000000"]}}
18:56:44 MQT: stat/TestBench/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
18:56:44 MQT: stat/TestBench/STATUS5 = {"StatusNET":{"Hostname":"TestBench-8169","IPAddress":"192.168.0.80","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.250","Mac":"DC:4F:22:91:FF:E9","Webserver":2,"WifiConfig":4}}
18:56:44 MQT: stat/TestBench/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.60","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_91FFE9","MqttUser":"admin","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
18:56:44 MQT: stat/TestBench/STATUS7 = {"StatusTIM":{"UTC":"Tue Mar 12 17:56:44 2019","Local":"Tue Mar 12 18:56:44 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:10","Sunset":"18:49"}}
18:56:44 MQT: stat/TestBench/STATUS10 = {"StatusSNS":{"Time":"2019-03-12T18:56:44"}}
18:56:44 MQT: stat/TestBench/STATUS11 = {"StatusSTS":{"Time":"2019-03-12T18:56:44","Uptime":"0T02:00:38","Vcc":3.209,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"ON","POWER3":"OFF","Wifi":{"AP":1,"SSId":"Byterite","BSSId":"18:E8:29:17:48:84","Channel":11,"RSSI":74}}}

```
- [x] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
18:47:27 MQT: stat/TestBench/RESULT = {"POWER3":"ON"}
18:47:27 MQT: stat/TestBench/POWER3 = ON

Hey Guys

OK, so I bought 3 each of sonoff T1 1CH, 2CH, 3CH touch switches.

I successfully flashed all 9 units and installed. All of the CH3 units are experiencing ghost switching. The 1CH and 2CH units are fine. 

I have replaced the 3CH units with 2CH units, I.E using the same wall box and switching the same lights (I was only using 2 lights on the 3CH anyway). The problem is not with the location of the units and not with the load they are switching as the 2CH unit is fine in the same wallbox. 

I erased and re-flashed all three 3CH units and put them on my test bench. Joined to wireless, but did not configure any MQTT broker, and they are all still randomly switching (every few hours any channel will switch on, never off) Also they beep a few times just after the channel switched on.

So my thought is that there is an issue with the 3CH units and tasmota? (they were not randomly switching using original firmware)

Has anyone else experienced this issue with the 3 CH T1 US units?

Thanks in Advance.

ascillato, 2019-03-12:
Please use weblog 4 for more debug information. When you use that, it will tell you in the console the source of the command that triggers your relay. It will be shown as SRC:

Bloodyagent, 2019-03-12:
ok, will enable weblog 4 and send result when random switching happens again, thanks

beppeserafini, 2019-03-12:
Hi, I have same issue. At the first time I thought was some interference on 315MHz receiver. So, seeing data sheet of SYN470R, I lift pin 11, in this way receiver is in shutdown. After that I put pin 10(DO line) at GND to avoid floating pin. After that I tried all my 7 sonoff without glass cover for 2 days and worked fine! no ghost switch.
So, I remount glass cover and ghost switching returns!
So, if you try to long press one of touch button you hear the first beep and after that quickly 2 beep. In original firmware with this operation you can pair with phone, with tasmota seems do nothing fortunately.
So, the problem is touch sensitivity. I try to find a solution but the are no name on IC between touch button and IC. I hope that someone know its name. I do reverse engineering on that and I draw schematic connection and I'm searching on web a datasheet with pin compatibility.
If is useful I can share my schematic on this part of circuit. If I have time tomorrow I'll see some signal coming from button with oscilloscope because maybe it's possibile to adjust sensitivity adjusting input RC network.

Bloodyagent, 2019-03-12:
Ok, just had random switching again, here is output

20:53:35 APP: Button2 immediate
20:53:35 SRC: Button
20:53:35 MQT: stat/TestBench/RESULT = {"POWER2":"ON"}
20:53:35 MQT: stat/TestBench/POWER2 = ON
20:53:35 CFG: Saved to flash at F4, Count 136, Bytes 3584

Thanks

Bloodyagent, 2019-03-12:
> Hi, I have same issue. At the first time I thought was some interference on 315MHz receiver. So, seeing data sheet of SYN470R, I lift pin 11, in this way receiver is in shutdown. After that I put pin 10(DO line) at GND to avoid floating pin. After that I tried all my 7 sonoff without glass cover for 2 days and worked fine! no ghost switch.
> So, I remount glass cover and ghost switching returns!
> So, if you try to long press one of touch button you hear the first beep and after that quickly 2 beep. In original firmware with this operation you can pair with phone, with tasmota seems do nothing fortunately.
> So, the problem is touch sensitivity. I try to find a solution but the are no name on IC between touch button and IC. I hope that someone know its name. I do reverse engineering on that and I draw schematic connection and I'm searching on web a datasheet with pin compatibility.
> If is useful I can share my schematic on this part of circuit. If I have time tomorrow I'll see some signal coming from button with oscilloscope because maybe it's possible to adjust sensitivity adjusting input RC network.

Thanks for this, I suspected this and did remove the cover at some point, but didn't really take note, think you may be right in that id didn't switch while cover was off. I didn't pursue this because its only happening with the 3CH versions and was an issue with original firmware. I may just disable the middle button and turn the 3CH into a 2CH to see if this fixes it.

Bloodyagent, 2019-03-12:
Just not sure why, if its a sensitivity problem, is it a non issue with original firmware and with the 2ch and 1ch versions?

beppeserafini, 2019-03-12:
I have also another sonoff t1 and I use it for 1 month without any ghost switch using last Tasmota firmware with the last core version. When I received others sonoff I saw sono disconnection and looking on the web I installed last Tasmota with 2.30 core and disconnections disappears. Some minutes ago sonoff that worked for a month had a ghost switch. So, now I flash last core version again and I want to see if ghost switch disapper. I’ll inform you

Bloodyagent, 2019-03-12:
Yes I moved to core 2.3.0 because of disconnects and haven't tested on 2.4.0

I've removed the white rectangular pads on switches....still getting ghost switching!

Let me know. Thanks

beppeserafini, 2019-03-13:
Did you remove all rectangular pads and did you have ghost switch without glass?

Bloodyagent, 2019-03-13:
Removed pads. Still ghost switching with cover on. Cover off no switching pads or no pads

Bloodyagent, 2019-03-13:
Ok so have compared the boards off the 1 ch and 3 ch switches. They are identical in every way. There are no missing components on the 1ch as opposed to the 3 channel. Only difference I can see is relay count. So why is it the that when I configure the 3ch as a 2ch the bottom touch button is disabled and not the middle similar to the 2ch. Was hoping to make the 3ch exactly as the 2ch to see if removing the middle touch switch would stop the ghost switching.

beppeserafini, 2019-03-13:
Can you see if in 1ch and 2ch versione there are some mark on ic touch component? In 3ch there is not mark on this. 
Thank you!
[image]

Bloodyagent, 2019-03-13:
Chips blank ok all boards. I Googled and there are a few 16pin touch chips out there. Looks to be a 4 ch touch chip

beppeserafini, 2019-03-13:
Today I had one ghost switch on button 2 with core 2.42. Do you have ghost switch only on middle button or also others? Now I try to put a paper sheet between cover and plastic pad. I'll see...
I found some chip but no one that I found is compatible with schematic I drawn by inspection(I think is correct).
If you found component with 16 pin that can fit please tell me. Or if you write down a list I can check if they can be fit.
[image]

ToRvaLDz, 2019-03-14:
I'm using about 20 Sonoff T1-2-3 in my home since one year with Tasmota and I found that ghost switching only happen with Mqtt retain and/or RF interference. 
In my case I solved disabling Mqtt retain, also try wifi config 4 or 5.

Bloodyagent, 2019-03-14:
Hi @ToRvaLDz 

This is on the test bench without mqtt configured. Only happens on my 3CH units, all of them.

I erased the flash, reflashed with 6.4.1 - 2.3.0, joined to wireless, chose sonoff T1 3CH from the config menu, restarted and left them on the testbench with no load and no mqtt, they all ghost switching.

Bloodyagent, 2019-03-14:
haven't tried wifi config 4 or 5. Will try...thanks

ToRvaLDz, 2019-03-14:
@Bloodyagent without Mqtt configured I had another issue, the touch is not responding every time ( look at thie video https://youtu.be/TUrKbRcctf8 I opened another issue for that #5461), also I noticed that if mqtt is not available it restart itself after some time, this could probably make a relay toggle.

beppeserafini, 2019-03-14:
@ToRvaLDz ghost switching you had like us? So, beep and then other two beep quickly?

ToRvaLDz, 2019-03-14:
@beppeserafini The beep indicate the long press, the only thing I understood is that the touch is not responding while tasmota is looking for an mqtt connection if not available and that if wifi is not connected tasmota do a reboot causing a relay toggle. For ghost switching, that is a button/relay that activate/deactivate itself, the only cause I found are: mqtt retain or wificonfig != 3 or 4.

Bloodyagent, 2019-03-14:
So its definitely interference of some type. Maybe a lowpass filter on the input pins of the IC? Not sure how the capacitive touch IC works? I see there is a space right next to the 1k resistor for a capacitor, not sure if this is meant to be an RC filter of some type?
[image]

Bloodyagent, 2019-03-14:
[image]
This looks close?

ToRvaLDz, 2019-03-14:
Just a stupid question: are you trying it with a serial console connected?

Bloodyagent, 2019-03-14:
Nope. Can we get more debug data that way?

ToRvaLDz, 2019-03-14:
No, just because I noticed a ghost switching issue while it is not mounted in the wall socket

beppeserafini, 2019-03-14:
@Bloodyagent component is compatible with schematic I drawn! I never seen bad data sheet like this!
There are only one thing that seems wrong with it. So, since pin1,2 are connected to gnd, the outputs behavior should be like this:
[image]
but in this way seems impossible to recognize long press because outputs toggle only the button is pressed again. There are others configurations that permit to detect long press but maybe is inverted in datasheet. I have to check this with oscilloscope.
However, looking datasheet, since description of detection is not described(but I think to know that), there is a capacitor that determines touch sensitivity. It's capacitor that I also drawn on pin 7. Datasheet suggests value from 4.7nF to 22nF. Translating from Chinese seems we have to reduce it in order to reduce sensitivity. If I have time tomorrow unmount cap from board and try to measure and substitute it with smaller value.

Bloodyagent, 2019-03-14:
All chinese documentation is bad🤣 Either way dude you a genius. I'm very rusty with digital electronics, I dont even own a scope anymore😫

Bloodyagent, 2019-03-14:
Pin7 is chip select? Not sure how that's going to affect touch sensitivity?

Bloodyagent, 2019-03-14:
Its C6? Ok just measured it. I get 22nf.

Bloodyagent, 2019-03-14:
Ok. I replaced it with a 4.7nf (That's all I have at home at the moment) Its reduced sensitivity to the point now where it only works with cover off. So looks like you are right!! I may just have to increase  it to 10nf, so will have to wait for tomorrow. For now I will leave it on testbench overnight. Lets see if it fixes the ghost issue!!

AndyGoldie, 2019-03-14:
I have been using 1 & 2 channel Sonoff switches for over a year without any ghost switching problems. I was using various versions of Tasmota so felt I should update all to 6.4.1. Big mistake now all 1 & 2 channel switches have random ghost switching. Therefore I strongly suspect this is some form of bug in 6.4.1.

Bloodyagent, 2019-03-14:
Hi Andy
I not have any issue with 1 and 2 ch. Just the 3ch. @beppeserafini seems to have nailed it with the touch sensitivity cap. So if you handy with a solder iron you can try change to a lower value. We still trying to figure out the correct value. I will also go back to an older firmware and try that. Thanks!

beppeserafini, 2019-03-14:
Yes I am. I design electronic boards daily. 😄 
Until now, since I put paper sheet between pad and cover, I didn't get ghost switches from that Sonoff. Doing that mean increase dielectric material and increase distance between "capacitor" armors, that would be decrease button sensitivity. So I think if it's mounted 22nF, a value about 15nF or 18nF could be good for us. 

_Pin 7 isn't chip select. It is called Csel. It's not explained how it works but generally pin like work like a constant time (series resistor inside) and it is compared with signals measured on pad pin._

beppeserafini, 2019-03-14:
> Its C6? Ok just measured it. I get 22nf. Which would agree with your theory.

Yes it is

ascillato, 2019-03-14:
@beppeserafini

>Until now, since I put paper sheet between pad and cover, I didn't get ghost switches from that Sonoff. Doing that mean increase dielectric material and increase distance between "capacitor" armors, that would be decrease button sensitivity.

Nice finding :+1:

Bloodyagent, 2019-03-15:
I walked into my study this morning and saw that the modified sonoff hadn't switched all night! So thought wow, its fixed. But then as I was about to start typing this it switched. I don't know if its because I'm sitting right here next to it and its picking up my body (lol) or if its because I placed my cell phone right next to it. I will get some higher value caps and try them with the units in the wall and see if its resolved. Will also try the paper idea. Anyway thanks for the help and assistance. This has been quite educational! Either way it seems we have 2 options now to resolve, and its clearly a hardware issue so I am going to close this now.

Bloodyagent, 2019-03-15:
adios amigos:)

simoneluconi, 2019-04-01:
Can someone explain properly where to find theese capacitor in the board and possible values to use to lower the sensitivity? We have about 30 Sonoff and about 4 of them suffer from ghost switching, it's really annoing. It is quite strange that those who suffer from this problem are positioned in the bathrooms.

beppeserafini, 2019-04-01:
It is C6 capacitor (0603 case). You can replace it with value near to 16nF. I mount 16,4nF with no sensitivity degradation but no ghost switching.

Bloodyagent, 2019-04-01:
I can also confirm this has resolved the ghosting issue.

beppeserafini, 2019-04-01:
Are you sure that ghost switching you have is not about RF interference? In this case you can follow also my solution in first reply I did

simoneluconi, 2019-04-01:
How could it be RF Interference? If no remote is programmed the switch couldn't switch by itself.

beppeserafini, 2019-04-01:
If you look on web you can see some video showing this issue on others sonoff products that are using same RF chip.

simoneluconi, 2019-04-03:
Should also point out that for that i can ear the sonoff also beeps, so it detect a long press on the button, not some "ghost press" or i believe "RF Interference"

beppeserafini, 2019-04-03:
If you ear beeps it's a sensitivity problem that you can resolve changing capacitor. I mount on my sonoff devices two 8.2nF ceramic capacitors in parallels (mount the second on top of the first)

xtremedays, 2019-04-26:
any other thoughts? I have ghosting issues on a 3ch T1 US, have put two 8.2nf ceramic caps on there, still having ghosting issues.

beppeserafini, 2019-04-26:
You can try with lower cap value, I.e. 15nF.
If you think is a RF interference and you aren’t using RF remote controller you can follow pcb modifications that I introduced in my first post https://github.com/arendst/Sonoff-Tasmota/issues/5449#issuecomment-472136413

xtremedays, 2019-04-26:
i've removed the RF chip, so doubt its that. Does it have to be a ceramic capacitor?

beppeserafini, 2019-04-26:
Yes, datasheet suggest X5R if I remember correctly.

HobbytronicsPK, 2019-05-13:
Have a look on to [Ghost / Random Switching on Sonoff Devices (MQTT Retained)](https://blog.hobbytronics.pk/sonoff-tasmota-tips-tricks/#Ghost-Random-Switching)

REALogics, 2019-05-23:
Just curious where this has left off.  It appears by commends here that CAPs did not necessarily fix the problem, so I have not tried it.  I did followed the MQTT Retained instructions in link above, but I still get ghost switching on both of my T1 3CH. The T1 starts with beep.... beep.beep.  Then the light changes state.  It is most often the center switch, but the top and bot switch also trigger on occasion.

REALogics, 2019-05-25:
With log turned, the ghost switched happens here:

`02:27:03 WIF: Checking connection...
02:27:03 WIF: Connected
02:27:05 APP: Button1 multi-press 1
02:27:16 SRC: Button
02:27:16 MQT: sonoff-5682/tele/STATE = {"Time":"2019-05-25T02:27:16","Uptime":"10T01:17:00","Vcc":3.202,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"ON","POWER3":"ON","Wifi":{"AP":1,"SSId":"XXXX","BSSId":"A0:63:91:D7:XX:XX","Channel":9,"RSSI":78,"LinkCount":2,"Downtime":"0T00:02:05"}}
02:27:16 MQT: sonoff-5682/stat/RESULT = {"POWER1":"OFF"}
02:27:16 MQT: sonoff-5682/stat/POWER1 = OFF (retained)
02:27:17 SRC: MQTT
02:27:17 RSL: Received Topic sonoff-5682/cmnd/POWER1, Data Size 3, Data OFF
02:27:17 RSL: Group 0, Index 1, Command POWER, Data OFF
02:27:17 MQT: sonoff-5682/tele/STATE = {"Time":"2019-05-25T02:27:17","Uptime":"10T01:17:01","Vcc":3.186,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"ON","POWER3":"ON","Wifi":{"AP":1,"SSId":"XXXX","BSSId":"A0:63:91:D7:XX:XX","Channel":9,"RSSI":72,"LinkCount":2,"Downtime":"0T00:02:05"}}
02:27:17 MQT: sonoff-5682/stat/RESULT = {"POWER1":"OFF"}
02:27:17 MQT: sonoff-5682/stat/POWER1 = OFF (retained)
02:27:17 CFG: Saved to flash at F4, Count 304, Bytes 3584
02:27:23 WIF: Checking connection...
02:27:23 WIF: Connected'

beppeserafini, 2019-05-25:
Did you try to keep sonoff without glass cover?
In my opinion is a sensitivity problem about touch

REALogics, 2019-05-28:
I was trying to avoid modifying hardware, thinking it could be done in firmware.  After looking at the chip document, it appears WTC6104 chip sensitivity is set by a capacitor (unlike the MPR121 set by firmware).  Since we are all putting tasmota firmware on these devices, the next question for me is... does this happen with Sonoff T1's with stock factory binary.  I don't have one to test.  But if stock binary does not have the issue, then there may still be something about tasmota firmware causing this.  I am studying the tasmota firmware at the moment.

beppeserafini, 2019-05-28:
Me too when the problem appeared I suspected that was a firmware bug but I didn’t have anymore sonoff to try. After that I saw no ghost switching without glass cover and I looked for hardware bugs..

psiho, 2019-05-29:
Im using T1 3 gang usa for 2 weeks. 11 days with stock FW and last 2 days with Tasmota. I havent noticed this problem on stock and noticed it 3 times in last 2 days with tasmota.

I have configured OpenHab to turn on some other lights on long press so even if i dont hear beep, pause, beep beep, I can see in the morning which channel ghosted a long press.

There's always a chance this is only accident, that somehow i missed ghost clicks on the stock fw, but I'd say 11 days (0) to 2 days (3)... Slim chance. I bet on some tasmota error or at least hardware error byoassed in original fw and not in tasmota.

sorokaalex, 2019-05-31:
This is my scenario. 

T1 1 gang: 3 un
T1 3 gang: 1 un

One T1 1 gang I’m running esp easy firmware. Others T1’s I’m running Tasmota firmware. 

First T1 1 gang I improved were flashed with esp easy firmware and I got ghost switch at few hours later. 

When received other T1’s I was deicied flash with Tasmota. 

T1’s 1 gang running last Tasmota firmware start ghost switch after few hours later also. 

As I had two T1 1 gang with different firmwares version with ghost switching, I was deciding test T1 3 gang with stock firmware and leave it for two weeks, results no ghost switching. After two weeks, flashed same T1 3 gang with Tasmota, result ghost switch appear again. 

After that, I was deicied running all T1’s 1 and 3 gang without glass cover, no ghost switching for three days tested. After three days, remount glass cover but now with an different assembly approach leaving glass cover for last one mounting it very carefully , results no ghost switching for few tested days, including T1 running esp easy firmware. 

All T1’s running Tasmota firmware I had pasted below commands at console

SwitchRetain Off
ButtonRetain On
ButtonRetain Off
Powerretain On

Part a time I thinked could be touch sensibility but when running with stock firmware no ghost switch appear.

AndyGoldie, 2019-05-31:
Hi

What version of Tasmota are you running?

I had the ghost switching problem with version 6.5

Once I downgraded the version to 6.3 all ghost problems stop with no modification to the devices required.

Best regards

Andy Goldie

sorokaalex, 2019-05-31:
> Hi What version of Tasmota are you running? I had the ghost switching problem with version 6.5 Once I downgraded the version to 6.3 all ghost problems stop with no modification to the devices required. Best regards Andy Goldie

I’m running 6.5 firmware on all Tasmota devices. If one more ghost switch appear I’ll try downgrade firmware version.

meingraham, 2019-06-01:
@AndyGoldie 

May I request that you try your tests with 6.5.0.13 available at http://thehackbox.org/tasmota/020300/sonoff.bin??

If your results are similar, please open a new issue providing the details of your stock firmware tests, Tasmota 6.3, and Tasmota 6.5.0.13..  Perhaps these clues will result in Theo being able to uncover what causes the instability.

Thanks

Mike

[272 more replies]

closed 2019-06-20 by ascillato2