arendst/Tasmota #5450: SonOff CH4 R2 whit BME680

Valerio1975, 2019-03-12:
I have a problem, I set everything for BME680 but I don't see the parameters. do i need to enable anything TX=GPIO1-> SDA
RX=GPIO3-> SCL

[image]

[image]

[image]

joba-1, 2019-03-12:
you use serial lines for i2c. Have you deactivated serial output?
The bme should be discovered and values displayed automatically.

ascillato, 2019-03-12:
Hi, please update to latest Tasmota version and try again. Thanks

Valerio1975, 2019-03-12:
How can I disable the serial port? I have the latest version.

⁣Inviato da TypeApp ​

ascillato, 2019-03-12:
With that you have the serial log disabled. Anyway, 6.4.1.9 is not the latest. Please update. Several fixes related to your issue has been applied after that.

After you update, please complete the troubleshooting template sharing also the output of status 0 command and the output of the console. Thanks

Valerio1975, 2019-03-13:
Sorry you could send me the complete link to update the latest version, because I did the update but this is the latest version. I also put the normal one in English (thinking that it changed from the Italian one) but nothing changes always remains the same. I put the minimal first and then sonoff.bin

ascillato2, 2019-03-13:
http://thehackbox.org/tasmota/sonoff.bin

Please, complete the troubleshooting template

closed 2019-03-13 by ascillato2

Valerio1975, 2019-03-13:
Thanks, too bad the update doesn't work. Ne upload of minimal ne sonoff.bin.
I will try to make an erase of the flash and program directly with the file you sent me the link.
Unfortunately, as in the old post you finally told me to use the chat, the update will not work either via internet or by hand. Once in the chat he offered one who told me he gave me a hand and then he disappeared, others tell me all about putting the minimal first and then the file I want. But when I say that it doesn't work, nobody answers anymore. Thanks a lot anyway.

Valerio1975, 2019-03-13:
I reprogrammed everything, now it works but putting sonoff-sensor.bin.
However, updating via OTP or manual still does not work.

[image]