arendst/Tasmota #5451: Controlling SonOff pow r2 external switch with a relay

asadulhuq, 2019-03-13:
I want a SonOff Pow r2 device to auto off after a predetermined on time like 1 hour. My purpose is to auto off a water heater after 1 hour or so.

I requested this feature for long time for Pow version (BTW, Inching feature is not available with this pow. It is a shame for Pow design team. They can off a device after 2 years but they cannot add a simple timer feature that unconditionally turns off a device after a short duration!)

Now I am thinking to make my own timer with a 4060 and a monostable circuit with 555. I hope my new system will apply a short circuit across the external push button of the Pow device. (A relay will be activated for 2 sec with the help of 555 monostable and apply a short circuit across the push button. Again, the 555 monostable circuit will be triggered by the 4060. The CD4060 timer will start counting time once the Pow will be turned on with the external switch. Thus the Sonoff will turn on the water heater and remain on for a duration, then off it after a predetermined time (eg. 1 hour) automatically and unconditionally.

How is my idea? Any suggestion is appreciated. Thanks in advance.

Frogmore42, 2019-03-13:
Look at the wiki
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands
Pulse time

You can also easily do this with most Home Automation software.  You do not need any extra hardware.

asadulhuq, 2019-03-13:
Frogmore42, Thanks for quick response. I have gone through the link you mentioned in your response. Could you please throw a couple of lines so that I can understand things. Please note I am not sure about a specific home automation software which could solve my problem. Thanks again.

ascillato2, 2019-03-13:
Hi

You just need to go to the Tasmota console and type `pulsetime 3600`

If you need further assistant, please find us in the Tasmota support chat. Github issues are for bug reporting and with the template completed. Thanks. 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-03-13 by ascillato2

asadulhuq, 2019-03-13:
> Hi
>
> You just need to go to the Tasmota console and type `pulsetime 3600`
>
> If you need further assistant, please find us in the Tasmota support chat. Github issues are for bug reporting and with the [...]

Is this okay for Sonoff Pow r2? I am afraid this will work for one time only! Thanks.

ascillato, 2019-03-13:
Yes, it is ok also for sonoff pow R2. Pulsetime command will work using Tasmota on any device that have any type of output