arendst/Tasmota #5454: Publish relay as "Lock" in Home Assistant

engrbm87, 2019-03-13:
I know that it is possible to enforce the sonoff relays as light switches when announced in Home assistant.

It would be great to have an option to announce the relay as as lock. I think we just need to change the config topic to /lock/unique_id/config. And we also need to change the topics for payload_on and payload_off to payload_lock and payload_unlock.

ascillato2, 2019-03-13:
Hi,

Thanks for sharing your ideas. 

This would not be easy due to some locks just need to receive a pulse with the relays to open and they close automatically and others use the same relay status to stay open or closed. So, all the cases should be taken into account adding more code to Tasmota.

Which one is your case?

engrbm87, 2019-03-13:
I think this can be managed by home assistance. if you try to configure a lock in HA it will either be locked or unlocked. HA offers a lock template that uses a secondary sensor for the door status.

The only thing we need to look at is whether LOCK means ON or OFF because some locks need power to turn on others unlock by turning power off.
In my case i have a lock that UNLOCkS by turning the relay ON and locks again once the relay is OFF.

ascillato, 2019-03-13:
>In my case i have a lock that UNLOCkS by turning the relay ON and locks again once the relay is OFF.

Ok. So, the logic of what it is needed to send to HA and the real status of the relay are different. So, it is not an easy fix like just anouncing to HA as a lock. Much more code is needed to achieve that.

One approach to reach a solution could be using the rules feature of Tasmota. It is not going to be an easy set of rules but can be achieved.

If you don't want to use rules, you can code the changes and make a pull request or wait to see if someone with a lock wants to make the changes.

Anyway, your idea is nice. Thanks for share.

engrbm87, 2019-03-14:
I have already done tome code changes and added the lock functionality. I am not an expert in coding but I managed to make setoption63 to enforce discovery as lock and added HASS_DISCOVER_LOCK[].
perhaps i can make a pull request and see if we can fine tune the code changes before merging.

ascillato2, 2019-03-14:
Great, please do it. Thanks

closed 2019-03-14 by ascillato2