arendst/Tasmota #5455: OBI Socket 2 Module Power LED not work without MQTT connection

Medel, 2019-03-13:
i'm using latest tasmota devel on obi socket 2.

The red Power LED (GPIO 13) does not work correctly. It shows the status of the relay only when MQTT is connected. It should work also with and without a connection (WIFI and MQTT)

Jason2866, 2019-03-13:
It seems there are 2 different OBI sockets model 2.
Try Template` {"NAME":"OBI 2-1","GPIO":[0,0,0,0,21,17,0,0,52,53,0,0,0],"FLAG":0,"BASE":61}`

ascillato2, 2019-03-13:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-03-13 by ascillato2

nagyrobi, 2019-05-11:
Tried this template. It does make the Power LED (red) operate correctly, however it makes the green LED light all the time, not flashing Tasmota-like.

meingraham, 2019-05-11:
Please visit [Chat](https://discord.gg/Ks2Kzd4) for more user experience. Now that it appears that LED assignments are identified, getting them to indicate as desired seems attainable. Troubleshooting your configuration here is not effective as there are many questions an tests to attempt.