arendst/Tasmota #5457: Tasmota on Tuya Smart Wireless WiFi Water Gas Shut-Off Valve

antibill51, 2019-03-13:
Hi.
Has anyone ever tried to install tasmota on a device like this one?
[https://www.alibaba.com/product-detail/Tuya-Smart-Wireless-WiFi-Water-Gas_60750642284.html](url)
Thanks.

Jason2866, 2019-03-13:
This is Github Issue. See https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md
Visit Tasmota Discord Chat for questions around Tasmota

ascillato2, 2019-03-13:
### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-03-13 by ascillato2

amitubale, 2019-06-17:
https://blakadder.github.io/templates/jinvoo_SM-PW713.html

prankousky, 2019-07-28:
I am waiting for this product to be delivered. Can you please explain how to flash tasmota onto it (or rather, where to solder rx,tx,3.3v and gnd? )?

Dylantje, 2020-05-30:
Hi some has this valve working?

NorthernMan54, 2020-06-18:
@Dylantje It appears that someone has this working after flashing with Tasmota.  And they note that they used to tuya-convert

https://templates.blakadder.com/owfeel_EN71.html

mrmaximas, 2024-04-22:
@NorthernMan54 may be you have backup of original tuya fw from this valve?? i want to revert it, but lost my backup ;(

NorthernMan54, 2024-04-22:
@mrmaximas Sorry, no backup

prankousky, 2024-04-23:
Me neither, sorry @NorthernMan54