arendst/Tasmota #5459: Poweroff a relay after max time without internet connection

itajackass, 2019-03-13:
I don't know if there is already exist this function so i'll ask for it.
Is there a way to poweroff a relay of a sonoff 4ch pro after a max time (example 30 minutes) for safety reasons like irrigation and internet shutdown.
I'll explain my scenario:

With Home assistant I control 4 irrigation valves with sonoff 4ch pro.
When I send command trought Home Assistant automation to start irrigation, it open in sequence: valve 1 for 10 minutes --> then close valve 1 and open valve 2 for 10 minutes ----> then close valve 2 and open valve 3 ---> then close valve 3 and open valve 4 --> finally close valve 4.

But in a bad scenario where valve 1 is open by Home Assistant AND THEN for some reasons wifi fails, router fails..... the sonoff doesn't receive the input to close valve 1 to continue irrigation program. SO VALVE 1 STILL REMAIN OPENED "FOREVER"!!!

So my question is: is there an internal clock countdown, maybe that can be set via MQTT, to close a relay after the countdown go to 0? In this way I'll can send a MQTT command to set the countdown when I send command to open a valve.... so when i open a valve for 10 minutes, i also send a command to create a countdown from 12 minutes to 0. if my wifi is ok, after 10 minutes HA close valve 1 and open valve 2 (countdown reached 2 minutes remaining)....but if wifi fails, after 10 minutes HA can't close the valve and countdown can go from 12 to 0....if reach 0 then sonoff close yourself all relay!

Jason2866, 2019-03-13:
This is Github Issue. See https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md
Maybe `Pulsetime` is a solution or a rule designed for your needs
Visit Tasmota Discord Chat for questions and help around Tasmota

Frogmore42, 2019-03-13:
GitHub issues is not the place to ask questions.  Use the chat link or the other resources listed in the template that is part of all new issues.

PulseTime is described in the wiki, look at the commands.  It probably does what you want.  Investigate that first, try it out and see if it meets your needs.  If not then come back with details of what is not working correctly and fill out the template.

closed 2019-03-13 by ascillato2