arendst/Tasmota #5461: Sonoff Touch not respond without WiFi connection

ToRvaLDz, 2019-03-13:
I have 21 sonoff touch since 1 year everything is working well except that when wifi connection is not available, the touch is not responding well, I think because it is trying restarting itself also with WiFi config 4 or 5, so I need to touch the button between a restart to power on the light.

ascillato2, 2019-03-13:
Please, could you be so kind on completing the troubleshooting template in order to have information so as to properly help you. Thanks.

ascillato2, 2019-03-14:
When you have time, please, complete the troubleshooting template or use the Tasmota support chat. Thanks

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-03-14 by ascillato2

ToRvaLDz, 2019-03-14:
@ascillato2 post updated thank you

ascillato, 2019-03-14:
Have you tried the precompiled bins?

ToRvaLDz, 2019-03-14:
I tried with precompiled bins and I have the same issue, I found that the problem appear while it is  checking mqtt connection:

```
17:31:29 MQT: Attempting connection...
17:31:34 MQT: Connect failed to domox.local:1883, rc -2. Retry in 10 sec
```

Setting MqttRetry 300 the touch is not responding for 2-3 seconds only every 5 minutes.

ascillato, 2019-03-14:
The hung when checking for mqtt had happened before but using the old pubsubclient library. With the new library that hung does not happend. That it is why I ask you to test the precompiled bins because you might be using an old version of that library.

So, as you have tested the precompiled bin, that is not the case, so it seems to be a problem with mDNS. If you set the mqtt server as an IP instead of an address, you can also perceive this hung?