arendst/Tasmota #5462: Issue with WebUi when choosing WS2812 in module configuration

timota, 2019-03-13:
### BUG DESCRIPTION
After flashing latest 2 versions of Tasmota (sonoff-282744d.bin and sonoff-6c69b8d.bin) got issue with WS2812.
Module Sonoff Basic R1. 
After choosing WS2812 as GPIO14 in module configuration, and trying to manage strip(on/off, use slider in WebUI) - WEBUI become unresponsive. In console sometimes it states as lost wifi or nothing.
After reboot - it states as wrong AP password, even if you change it again via console

```
Backlog ssid1 SSID; password1 PASSWORD
```
it doesnt help. Only "reset 2" helps. Tried reflash couple times with differernt FW(above) - issue the same. 

### REQUESTED INFORMATION

```
STATUS 0 OUTPUT HERE:
23:48:54 CMD: status 0
23:48:54 SRC: Serial
23:48:54 RSL: Received Topic /status, Data Size 1, Data 0
23:48:54 RSL: Group 0, Index 1, Command STATUS, Data 0
23:48:54 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"sonoff","ButtonTopic":"0","Power":2,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:48:54 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:02:15","StartupUTC":"2019-03-13T22:46:39","Sleep":50,"CfgHolder":4617,"BootCount":3,"SaveCount":16,"SaveAddress":"F4000"}}
23:48:54 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.4.1.21(6c69b8d-sonoff)","BuildDateTime":"2019-03-11T20:01:47","Boot":31,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
23:48:54 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":5,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["VM1126584",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
23:48:54 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":552,"Free":448,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","0FDAE794","000783A0","23B617CE","00003BC0"]}}
23:48:54 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-6518","IPAddress":"(IP unset)","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"DC:4F:22:0A:D9:76","Webserver":2,"WifiConfig":4}}
23:48:54 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0AD976","MqttUser":"DVES_USER","MqttType":1,"MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
23:48:54 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Wed Mar 13 22:48:54 2019","Local":"Wed Mar 13 23:48:54 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:08","Sunset":"18:51"}}
23:48:54 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-03-13T23:48:54"}}
23:48:54 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-03-13T23:48:54","Uptime":"0T00:02:15","Vcc":3.304,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":56,"POWER1":"OFF","POWER2":"ON","Dimmer":47,"Color":"777777","HSBColor":"0,0,47","Channel":[46,46,46],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"VM1126584","BSSId":"B0:39:56:D6:C2:2C","Channel":6,"RSSI":90,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
00:00:04 HTP: Web server active on sonoff-6518 with IP address 192.168.1.176
00:00:04 APP: (UTC) Wed Mar 13 22:46:43 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
23:46:46 HTP: Main Menu
23:46:49 APP: Boot Count 3
23:46:49 CFG: Saved to flash at F9, Count 11, Bytes 3584
23:46:55 SRC: WebGui from 192.168.1.53
23:46:55 RSL: stat/sonoff/RESULT = {"POWER1":"ON"}
23:46:55 RSL: stat/sonoff/POWER1 = ON
23:46:55 CFG: Saved to flash at F8, Count 12, Bytes 3584
23:46:57 SRC: WebGui from 192.168.1.53
23:46:57 RSL: stat/sonoff/RESULT = {"POWER1":"OFF"}
23:46:57 RSL: stat/sonoff/POWER1 = OFF
23:46:57 CFG: Saved to flash at F7, Count 13, Bytes 3584
23:46:59 SRC: WebGui from 192.168.1.53
23:46:59 RSL: stat/sonoff/RESULT = {"POWER2":"ON"}
23:46:59 RSL: stat/sonoff/POWER2 = ON
23:47:00 CFG: Saved to flash at F6, Count 14, Bytes 3584
23:47:03 SRC: WebGui from 192.168.1.53
23:47:03 RSL: stat/sonoff/RESULT = {"POWER2":"OFF"}
23:47:03 RSL: stat/sonoff/POWER2 = OFF
23:47:03 CFG: Saved to flash at F5, Count 15, Bytes 3584
23:47:04 WIF: Checking connection...
23:47:04 WIF: Connected
23:47:09 SRC: WebGui from 192.168.1.53
23:47:09 RSL: Received Topic /Dimmer, Data Size 2, Data 47
23:47:09 RSL: Group 0, Index 1, Command DIMMER, Data 47
23:47:09 SRC: Light
23:47:09 RSL: stat/sonoff/RESULT = {"POWER2":"ON","Dimmer":47,"Color":"777777","HSBColor":"0,0,47","Channel":[46,46,46]}
23:47:09 CFG: Saved to flash at F4, Count 16, Bytes 3584
23:47:15 HTP: Console
23:47:24 WIF: Checking connection...
23:47:24 WIF: Connected
23:47:44 WIF: Checking connection...
23:47:44 WIF: Connected
23:48:04 WIF: Checking connection...
[... 113 lines cut ...]
00:00:07 WIF: Attempting connection...
00:00:08 WIF: Checking connection...
00:00:08 WIF: Attempting connection...
00:00:09 WIF: Checking connection...
00:00:09 WIF: Attempting connection...
00:00:10 APP: Boot Count 9
00:00:10 CFG: Saved to flash at F9, Count 19, Bytes 3584
00:00:10 WIF: Checking connection...
00:00:10 WIF: Attempting connection...
00:00:11 WIF: Checking connection...
00:00:11 WIF: Attempting connection...
00:00:12 WIF: Checking connection...
00:00:12 WIF: Connect failed with AP incorrect password
00:00:12 WIF: Connecting to AP1 VM1126584 in mode 11N as sonoff-6518...
00:00:13 WIF: Checking connection...
00:00:13 WIF: Attempting connection...
00:00:14 WIF: Checking connection...
00:00:14 WIF: Attempting connection...
00:00:15 WIF: Checking connection...
00:00:15 WIF: Attempting connection...
00:00:16 WIF: Checking connection...
00:00:16 WIF: Attempting connection...
```

Regards

Jason2866, 2019-03-14:
The LED Stripe is powered from a external power supply?
If not you overload the power supply from sonoff device which results in issues...
For support visit us in Discord Chat.

timota, 2019-03-14:
Hi, 

actually not a stripe but simillar ring [https://hobbycomponents.com/displays/966-66mm-ws2812b-24-rgb-led-ring](url)

and yes it powered not from external power supply - directly from sonoff. 
but more strange - it works for ages before (started from tasmota 5.xxx, including 6.4.0). But i decided to upgrade FW to 6.4.1 and got issue. So dont think that this is related to PS, untill something changed in code to manage power.

Reagards

ascillato, 2019-03-14:
>yes it powered not from external power supply - directly from sonoff

Sorry. That is the issue. Wifi requires some spikes of current and the sonoff basic is known to have a very bad power regulator. 

So, you will need to add an external power source. 

The new Tasmota versions use a little more power due more code is being executed for all the new features added. Also, if you use the power regulator at its limit, it will degrade faster showing these failures.

ascillato2, 2019-03-14:
Closing this issue as it is hardware related. Thanks for all the testings and comments.

closed 2019-03-14 by ascillato2

timota, 2019-03-14:
ok - thanks, i will test with external PS, and come back if this will not resolve issue. 

Thank you.