arendst/Tasmota #5463: Telemetry data switch state from Tasmota has no effect on Home Assistant, causes loss of sync after HA restart

melikyuksel, 2019-03-14:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION
The [**Home Assistant ‐ Manual Config** page](https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant-%E2%80%90-Manual-Config) that I'm using with Tasmota Basic has a section **Tip: Sync power state**, where it states "The device state will always be synced each TelePeriod (every 5 minutes by default)."

Indeed, I'm getting the `tele/kitchen-lights/STATE = {"Time": ...`information through to the MQTT broker Mosquitto, and it contains the POWER state (ON or OFF). However, the given `configuration.yaml` for Home Assistant example does not seem to actually provide the state information to the Home Assistant. Mosquitto is receiving the messages, but the state doesn't update in Home Assistant. So if HASS is restarted or Wi-Fi is lost, then the state of the Sonoff devices with Tasmota doesn't update to HASS because HASS doesn't understand the telemetry data, only the state data.

This may not be just a Tasmota issue strictly, but the setup information I used is purely from the Tasmota setup page I linked above, so I'm filing the issue here! Here is the example `configuration.yaml` section I used from that page:

```
# Example configuration.yaml entry
switch:
  - platform: mqtt
    name: "Sonoff power"
    state_topic: "stat/sonoff/RESULT"
    value_template: "{{ value_json.POWER }}"
    command_topic: "cmnd/sonoff/POWER"
    availability_topic: "tele/sonoff/LWT"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    payload_available: "Online"
    payload_not_available: "Offline"
    retain: false
```

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic): Sonoff Basic
- [x] Tasmota binary firmware version number used: 6.4.1 (pre-compiled)
- [x] Provide the output of command ``status 0``:
```
02:57:35 MQT: stat/kitchen-cabinet-lights/STATUS = {"Status":{"Module":1,"FriendlyName":["Kitchen Cabinet Lights"],"Topic":"kitchen-cabinet-lights","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
02:57:35 MQT: stat/kitchen-cabinet-lights/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"Software/System restart","Uptime":"0T15:39:58","StartupUTC":"2019-03-13T10:17:37","Sleep":50,"BootCount":4,"SaveCount":35,"SaveAddress":"F9000"}}
02:57:35 MQT: stat/kitchen-cabinet-lights/STATUS2 = {"StatusFWR":{"Version":"6.4.1(basic)","BuildDateTime":"2018-12-24T14:43:30","Boot":4,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
02:57:35 MQT: stat/kitchen-cabinet-lights/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["*widaz","NETGEAR"],"TelePeriod":300,"SetOption":["0000A029","558180C0","00000000"]}}
02:57:35 MQT: stat/kitchen-cabinet-lights/STATUS4 = {"StatusMEM":{"ProgramSize":445,"Free":556,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000809","0F082780","000083A0","21021706","000001C0"]}}
02:57:35 MQT: stat/kitchen-cabinet-lights/STATUS5 = {"StatusNET":{"Hostname":"kitchen-cabinet-lights","IPAddress":"10.11.12.138","Gateway":"10.11.12.1","Subnetmask":"255.255.255.0","DNSServer":"10.11.12.1","Mac":"3C:71:BF:25:6E:62","Webserver":2,"WifiConfig":4}}
02:57:35 MQT: stat/kitchen-cabinet-lights/STATUS6 = {"StatusMQT":{"MqttHost":"10.11.12.30","MqttPort":1883,"MqttClientMask":"kitchen-cabinet-lights_%06X","MqttClient":"kitchen-cabinet-lights_256E62","MqttUser":"user1","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
02:57:35 MQT: stat/kitchen-cabinet-lights/STATUS7 = {"StatusTIM":{"UTC":"Thu Mar 14 01:57:35 2019","Local":"Thu Mar 14 02:57:35 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:06","Sunset":"18:53"}}
02:57:35 MQT: stat/kitchen-cabinet-lights/STATUS10 = {"StatusSNS":{"Time":"2019-03-14T02:57:35"}}
02:57:35 MQT: stat/kitchen-cabinet-lights/STATUS11 = {"StatusSTS":{"Time":"2019-03-14T02:57:35","Uptime":"0T15:39:58","Vcc":3.445,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"mynetwork","BSSId":"xx:xx:xx:xx:xx:xx","Channel":10,"RSSI":78}}}
```
Console output every 5 minutes:
```
02:52:53 MQT: tele/kitchen-cabinet-lights/STATE = {"Time":"2019-03-14T02:52:53","Uptime":"0T15:35:16","Vcc":3.461,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"mynetwork","BSSId":"xx:xx:xx:xx:xx:xx","Channel":10,"RSSI":70}}
```

### TO REPRODUCE
1. Use the default, latest, fresh HASS install with local Mosquitto as MQTT broker
2. Follow the [Home Assistant ‐ Manual Config](https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant-%E2%80%90-Manual-Config) to add a Sonoff Tasmota switch.
3. Turn on the switch using the physical button.
4. Restart the Home Assistant service.
5. Look at the switch state on Home Assistant. It will be off, instead of the actual state of on.

### EXPECTED BEHAVIOUR
The documentation at the link should have instructions on how to integrate Tasmota devices into Home Assistant so that the "The device state will always be synced each TelePeriod (every 5 minutes by default)." thing actually works. Unless that sentence is not in fact how it's supposed to work, then it should be deleted, I guess?

Jason2866, 2019-03-14:
After a restarting HA, the states are not in sync. You have to take care for this. How is described in wiki https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant-%E2%80%90-Manual-Config#tip-sync-power-state

ascillato2, 2019-03-14:
Closing this issue as it is duplicated. 

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2019-03-14 by ascillato2

ascillato, 2019-03-15:
@melikyuksel 

Sorry, I have reverted your change in the wiki. The order of the topic was ok before your change.

Please, re-read the whole article.

If you use the manual approach, you have to see the other document where the order is like your change (https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant-%E2%80%90-Manual-Config)

If you use the automatic approach (as in the document you changed) the order is different.

If you have doubts, please ask in the Tasmota Chat BEFORE modifying the wiki. Thanks.

melikyuksel, 2019-03-19:
Thanks! I set up PowerRetain 1 and it did what I was looking for. However, the line "The device state will always be synced each TelePeriod (every 5 minutes by default)." is still not accurate AFAIK, since the every-5-minute telemetry data indeed includes power state but HASS has no way to interpret that for its state information from the way that the page describes things.

ascillato, 2019-03-19:
Correct. That sentence is not meant for the automation for _sync devices state at HA start_. I have fixed that article. Thanks.