arendst/Tasmota #5464: Support request: Disable Wifi fail indication on RGB devices?

sgofferj, 2019-03-14:
I recently got a new Arrilux LC-01 for my bedroom and I noticed that it sometimes over night turns red uncommanded. I know that my Sonoff B1 go red if they have some problem with the Wifi, so I assume, it's the same here.
Is it possible to switch this behavior off or modify it so that after reconnect, the device goes back to the state it was in before the disconnect? I went through the SetOption list in the Wiki but nothing seems to fit.

Jason2866, 2019-03-14:
No, there is no light change if you have wifi problems. So your color change is initiated somewhere else....

ascillato2, 2019-03-14:
Set weblog to 4 and check what the console told you about the source of that command. You will see SRC:

ascillato2, 2019-03-14:
Please, complete the troubleshooting template in order to have information so as to properly help you. Thanks

sgofferj, 2019-03-14:
@ascillato2 Where do I find the template?
@Jason2866 I checked from the console for a while and the light apparently goes red when it fails to connect to the MQTT broker, so wifi wasn't the direct cause. But it stays red after it successfully reconnects to the broker.

sgofferj, 2019-03-14:
Found the template

sgofferj, 2019-03-14:
### ISSUE DESCRIPTION - TROUBLESHOOTING
Arilux LC-01 module turns LEDs red every once in a while. Originally I suspected Wifi problems but after watching the log for a while I see it happening in connection with disconnts from the MQTT server which could be caused by Wifi issues.
  
### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
Not a member of Google groups for privacy reasons
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
Not a member of discord for privacy reasons
- [X] Device used (i.e. Sonoff Basic) : Arrilux LC-01
- [X] Tasmota binary firmware version number used : 6.4.1 / pre-compiled
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [X] Provide the output of command ``status 0`` :
```
13:40:27 CMD: status 0
13:40:27 SRC: WebConsole from 192.168.10.1
13:40:27 RSL: Received Topic /status, Data Size 1, Data 0
13:40:27 RSL: Group 0, Index 1, Command STATUS, Data 0
13:40:27 MQT: rgb_1/stat/STATUS = {"Status":{"Module":37,"FriendlyName":["RGB_1"],"Topic":"rgb_1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:40:27 MQT: rgb_1/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.10.201/api/arduino/full/sonoff-6.3.0.bin","RestartReason":"Software/System restart","Uptime":"0T01:54:39","StartupUTC":"2019-03-14T10:45:48","Sleep":0,"BootCount":20,"SaveCount":145,"SaveAddress":"FB000"}}
13:40:27 MQT: rgb_1/stat/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2018-12-24T14:41:00","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
13:40:27 MQT: rgb_1/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Stefan_Gofferje_IoT",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000400"]}}
13:40:27 MQT: rgb_1/stat/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE794","000383A0","23B617CE","00003BC0"]}}
13:40:27 MQT: rgb_1/stat/STATUS5 = {"StatusNET":{"Hostname":"rgb_1-2262","IPAddress":"192.168.2.20","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.10.201","Mac":"84:0D:8E:5C:E8:D6","Webserver":2,"WifiConfig":3}}
13:40:27 MQT: rgb_1/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.10.201","MqttPort":1883,"MqttClientMask":"RGB_1","MqttClient":"RGB_1","MqttUser":"machineuser","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
13:40:27 MQT: rgb_1/stat/STATUS7 = {"StatusTIM":{"UTC":"Thu Mar 14 12:40:27 2019","Local":"Thu Mar 14 13:40:27 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:06","Sunset":"18:53"}}
13:40:27 MQT: rgb_1/stat/STATUS10 = {"StatusSNS":{"Time":"2019-03-14T13:40:27"}}
13:40:27 MQT: rgb_1/stat/STATUS11 = {"StatusSTS":{"Time":"2019-03-14T13:40:27","Uptime":"0T01:54:39","Vcc":3.524,"SleepMode":"Normal","Sleep":0,"LoadAvg":168,"POWER":"OFF","Dimmer":100,"Color":"FF0000","HSBColor":"0,100,100","Channel":[100,0,0],"Scheme":0,"Fade":"ON","Speed":4,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Stefan_Gofferje_IoT","BSSId":"00:1D:A1:77:35:B1","Channel":1,"RSSI":72}}}
```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
Waiting for problem to occur at the moment. Log will come when I got it.
```
**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2019-03-14:
Please, add the last part where it asks about the console output when that happens. Remember to have weblog to 4. Thanks

ascillato2, 2019-03-14:
With that information we will address the issue of the red led turning on, but then we should address the other issue of mqtt disconnections. The mqtt issue can be solved as explained: https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#Wi-Fi-issues-arduino-core-versions-and-expressif-sdk

To force the mqtt disconnection to test the red led issue, please turn off your mqtt broker

sgofferj, 2019-03-14:
Did a restart to have a clean log. Module started with leds set to red although it was off before the restart.
More logs coming as issues occur.

`00:00:00 CFG: Loaded from flash at FA, Count 146
00:00:00 SRC: Restart
00:00:00 Project sonoff RGB_1 Version 6.4.1(sonoff)-2_4_2
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 Stefan_Gofferje_IoT in mode 11N as rgb_1-2262...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Connected
00:00:05 DNS: Initialized
00:00:05 HTP: Web server active on rgb_1-2262.local with IP address 192.168.2.20
00:00:05 APP: (UTC) Thu Mar 14 13:00:48 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
14:00:50 MQT: Attempting connection...
14:00:50 MQT: Connected
14:00:50 MQT: rgb_1/tele/LWT = Online (retained)
14:00:50 MQT: rgb_1/cmnd/POWER = 
14:00:50 MQT: Subscribe to rgb_1/cmnd/#
14:00:50 MQT: Subscribe to sonoffs/cmnd/#
14:00:50 MQT: Subscribe to cmnd/RGB_1_fb/#
14:00:50 MQT: rgb_1/tele/INFO1 = {"Module":"Arilux LC01","Version":"6.4.1(sonoff)","FallbackTopic":"cmnd/RGB_1_fb/","GroupTopic":"sonoffs"}
14:00:50 MQT: rgb_1/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"rgb_1-2262","IPAddress":"192.168.2.20"}
14:00:50 MQT: rgb_1/tele/INFO3 = {"RestartReason":"Software/System restart"}
14:00:50 MQT: rgb_1/stat/RESULT = {"POWER":"OFF"}
14:00:50 MQT: rgb_1/stat/POWER = OFF
14:00:50 HTP: Main Menu
14:00:52 HTP: Console
14:00:53 APP: Boot Count 21
14:00:53 CFG: Saved to flash at F9, Count 147, Bytes 3584
14:00:58 MQT: rgb_1/tele/STATE = {"Time":"2019-03-14T14:00:58","Uptime":"0T00:00:15","Vcc":3.526,"SleepMode":"Normal","Sleep":0,"LoadAvg":47,"POWER":"OFF","Dimmer":100,"Color":"FF0000","HSBColor":"0,100,100","Channel":[100,0,0],"Scheme":0,"Fade":"ON","Speed":4,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Stefan_Gofferje_IoT","BSSId":"00:1D:A1:77:35:B1","Channel":1,"RSSI":74}}
`

ascillato2, 2019-03-14:
To force the mqtt disconnection to test the red led issue, please turn off your mqtt broker and then post the output of the console. Thanks

sgofferj, 2019-03-15:
I can't reproduce the issue anymore at the moment. I configured logging via syslog to my logserver and will report back when I encounter the symptoms again. I guess, the ticket can be closed in the meanwhile.

closed 2019-03-15 by ascillato2