arendst/Tasmota #5465: The Hebrew Release not in Hebrew.

rt400, 2019-03-14:
Hi ,

The Hebrew bin file does not contain in hebrew. show only in english .

please fix it. thanks .

ascillato, 2019-03-14:
Hi,

The language file for hebrew is not completed translated.

If you are able to edit it and translate it, you will be helping us a lot.

ascillato, 2019-03-14:
If you want to help us, one way of editing the language file is browsing the code by the webpage of github, go to the folder language and open he-HE.h
Then click on the pencil (edit), make the modifications you think are appropiated and then you can save it. After that, github will save the changes on your own fork (your own version of Tasmota) and right after that will pop up a message to make a pull request to main Tasmota repository for verification.

ascillato2, 2019-03-14:
If you need further assistant, please ask here or in the Tasmota support chat. Thanks.

closed 2019-03-14 by ascillato2

rt400, 2019-03-14:
@ascillato it was translate by me. once..
This is not intentionally completed because there are no Hebrew words for English words so there are still English words in this file. Most of the things that can be translated logically made the rest do not need to be translated.