arendst/Tasmota #5466: Bootloop

AIexBV, 2019-03-14:
This rule leads to a bootloop instantly:

Rule1 on System#Boot var1 0 endon on Tele-Energy#Power>0 do Scale1 %var1%,0,100,0,100 endon

ascillato, 2019-03-14:
Please, could you be so kind on completing the troubleshooting template in order to have information so as to properly help you. Thanks.

ascillato2, 2019-03-14:
Sorry, I can't reproduce your issue. Tested in a sonoff pow R2 and no issue.
Please, complete the troubleshooting template. Thanks

netpok, 2019-03-14:
There is two problem with the rule above:
1. `System#Boot` happens after mqtt is initialized, you probably want to use `Power1#Boot` which happens after tasmota boot
2. Probably the reason of the bootloop is that you missed the `do` keyword here: `on System#Boot var1 0 endon`

AIexBV, 2019-03-14:
@netpok - While looking out to find the issue template, you have had the right idea!

closed 2019-03-14 by AIexBV