arendst/Tasmota #5473: Energy Calc after Power Cut

BearTi, 2019-03-15:
Hi,

I´ve a Gesund power socket with Tosmata and every time  I´ve a power failure in my appartment or when disconnect the socket from power, the Energy values are reseted.

The values are not 0, but are not either the values before the power cut.

What can I do?

Thanks

Riwaha, 2019-03-15:
It appears it is only saved if you restart the device with the restart button, otherwise it never saves the energy or counter info.
I believe this is to keep writes low to the device.

Perhaps a periodic save(configurable period) would work, this way it can keep the writes low while also keeping the data partially accurate.

Jason2866, 2019-03-15:
As designed. Measured values arent saved in realtime in flash. This is done from time to time
Why? to protect flash from getting corrupted from to much write cycles 
So if a power outtake is there you loose some data. 
If the data is urgent you have to collect the data in realtime in a Home Automation system

arendst, 2019-03-15:
> It appears it is only saved if you restart the device with the restart button, otherwise it never saves the energy or counter info.

Energy counter info is saved daily at 00:00

closed 2019-03-15 by ascillato2