arendst/Tasmota #5477: IRhvac LG problems.

daaf84, 2019-03-15:
<GUIDE>

<This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum>
  
<Please DO NOT OPEN AN ISSUE:>
  <If you have general questions or you need help on Tasmota usage, go to the Tasmota support chat>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a new device, please use the Tasmota Template Feature. See wiki for that>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### ISSUE DESCRIPTION - TROUBLESHOOTING
When sending 

irhvac { "Vendor": "LG", "Power": 1 } my LG AC turns on, thats good,
But hen I send: irhvac { "Vendor": "LG", "Fanspeed": "7"} it turns off...

if I try irhvac { "Vendor": "LG", "Power": 1, "Fanspeed": "7"} I get an error:
16:51:28 CMD: irhvac { "Vendor": "LG", "Power": 1, "Fanspeed": "7"}
16:51:28 MQT: sonoff_5/stat/RESULT = {"IRHVAC":"Wrong VENDOR, MODE or FANSPEED"}
  

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Sonoff Basic
- [x] Tasmota binary firmware version number used : 6.4.1 /  self-compiled
- [x] Development IDE - Compiler / Upload tools used : Atom Platform IO
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

17:01:59 CMD: status 0
17:01:59 MQT: sonoff_5/stat/STATUS = {"Status":{"Module":1,"FriendlyName":["SonoffTEST"],"Topic":"sonoff_5","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:01:59 MQT: sonoff_5/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:14:45","StartupUTC":"2019-03-15T15:47:14","Sleep":50,"BootCount":44,"SaveCount":101,"SaveAddress":"F7000"}}
17:01:59 MQT: sonoff_5/stat/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2019-03-15T16:31:24","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
17:01:59 MQT: sonoff_5/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["diefdijk",""],"TelePeriod":300,"SetOption":["60028009","558180C0","00000100"]}}
17:01:59 MQT: sonoff_5/stat/STATUS4 = {"StatusMEM":{"ProgramSize":511,"Free":492,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","0FDE6794","240383A0","23B617C6","000028C0"]}}
17:01:59 MQT: sonoff_5/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoff_5-0643","IPAddress":"192.168.1.173","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"5C:CF:7F:7C:A2:83","Webserver":2,"WifiConfig":5}}
17:01:59 MQT: sonoff_5/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7CA283","MqttUser":"daaf84","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
17:01:59 MQT: sonoff_5/stat/STATUS7 = {"StatusTIM":{"UTC":"Fri Mar 15 16:01:59 2019","Local":"Fri Mar 15 17:01:59 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:04","Sunset":"18:54"}}
17:01:59 MQT: sonoff_5/stat/STATUS10 = {"StatusSNS":{"Time":"2019-03-15T17:01:59"}}
17:01:59 MQT: sonoff_5/stat/STATUS11 = {"StatusSTS":{"Time":"2019-03-15T17:01:59","Uptime":"0T00:14:45","Vcc":3.423,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"diefdijk","BSSId":"00:24:36:A6:A3:C1","Channel":6,"RSSI":100}}}
17:02:00 MQT: sonoff_5/tele/UPTIME = {"Time":"2019-03-15T17:02:00","Uptime":"0T00:14:46"}
```
- [x] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
17:05:24 CMD: weblog 4
17:05:24 SRC: WebConsole from 192.168.1.13
17:05:24 RSL: Received Topic /weblog, Data Size 1, Data 4
17:05:24 RSL: Group 0, Index 1, Command WEBLOG, Data 4
17:05:24 MQT: sonoff_5/stat/RESULT = {"WebLog":4}
17:05:40 WIF: Checking connection...
17:05:40 WIF: Connected

```
**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-03-15:
Try the demo sketch from the used library if your AC is supported 
https://github.com/arendst/Sonoff-Tasmota/tree/development/lib/IRremoteESP8266-2.5.2.03

ascillato2, 2019-03-15:
Available options:

"FanSpeed":"1"|"2"|"3"|"4"|"5"|"Auto"|"Silence"

closed 2019-03-15 by ascillato2