arendst/Tasmota #5478: FEATURE Request: Return GPIO14 status instead of Relay Status for Wemo Emulation

Bart56, 2019-03-15:
I use the Wemo emulation specifically for Alexa native operation and to avoid going out to the cloud. I have used fauxmoESP code on a NodeMCU to natively control my garage door. Once the wemo ON command is received, the relay toggles on for 1/2 second. When the reed switch on the door opens, the wemo return status is  the open reed switch. However, after using Tasmota on a sonoff basic for some lights, I thought this would be a perfect platform for a garage door opener solution. So I'm trying...

The only problem I've run into, is that after setting pulsetime so the relay is on for half a second then goes off, Alexa sees the status of the wemo device as off after the garage door is open. A reed switch on GPIO14 set to Switch1 allows remote operation of the relay and I saw in another thread that we can completely decouple GPIO14 from the relay but (and here is the problem)... it only reports GPIO14 status through MQTT.  Alexa doesn't do MQTT natively.

 I've looked through the code and see where wemo status is returned, but I would like to return GPIO14 status instead of relay status in the wemo GetBinaryState. In this way, if the garage door reed switch is open after turning on the relay momentarily, Alexa would show a status of ON as long as the door is open. 

The bad news is I don't have enough experience to modify this code but will continue to research how to compile it after changing it to use GPIO14.  It would be really nice to include it into one of the switchmode commands. If anyone can point me in the right direction, I would be grateful.

ascillato, 2019-03-15:
Hi,

Sorry. Your request is very specific and only to your setup. The way Tasmota was develop is to be used with a home automation software. You should try setting up one as referenced in the wiki.

We will be glad on helping you. You can find us in the Tasmota support chat. Thanks.

closed 2019-03-15 by ascillato2

ascillato, 2019-03-15:
Anyway, Alexa alone uses the cloud a lot. All the audio that it records is being sent to amazon servers. Using the Emulation in Tasmota will not avoid Alexa using the could. The only way to 100% avoiding the cloud is setting a home automation software like openhab or home assistant and not using Alexa.

Bart56, 2019-03-15:
Thanks for the quick response @ascillato. 
Fair point about Alexa using the cloud anyway,  But every add-on smart home device seems to want credentials for their own cloud service. I personally do have home assistant running which is a great platform, but this is for folks (relatives really) who don't need or want a hub or spend hours of time configuring a home automation platform (nor IFTTT, MQTT Broker, etc) just to control their garage.

I am using the ardiuno IDE to try to compile sonoff.ino and of course am hitting all the fatal compilation errors one-by one.  So I'll head over to the Support Chat to get some assistance in getting the unmodified sonoff.ino to compile. After that, I realize I'll be on my own.  I was hoping someone had already done something similar and could point me in the right direction.

Again, many thanks for the quick response.

Bart56, 2019-03-15:
> You can find us in the Tasmota support chat.

ummm... where do I find the support chat?

ascillato, 2019-03-15:
https://discord.gg/Ks2Kzd4

Bart56, 2019-03-17:
Thank you Adrian.
I have been able to compile the tasmota source and change the wemo
getbinarystate to return that of gpio14. I am having some difficulties
navigating the code to do much better. I would like to see the status of
gpio14 on the web interface main page, but I would need a long time to
dissect all the code modules. It is a crude solution and I'm unsure if the
forum can help me understand how it all works and what I need to change to
get the status on the root web page.  Thank you again.
Bart

ascillato, 2019-03-17:
>I would like to see the status of gpio14 on the web interface main page

You can make a rule to display that in the console or in your home automation.

Bart56, 2019-03-18:
Right although I'm trying to deploy this to only use Alexa because of the
native wemo support. My daughter won't have use for the console and doesn't
have a home automation system. It was mostly for me to test just by looking
at the main web page or by asking her what the main web page says if she
calls me on the phone with an issue.
So I looked at rules here:
https://github.com/arendst/Sonoff-Tasmota/wiki/Rules but didn't find
anything to change the source of the ON and OFF status of the relay to be
status of the switch. I'm probably being too dense or just not seeing it. I
also haven't found where the code is that display's the relay status there.
I see where the page is built by finding "div id='l1'  in
xdrv_01_webserver.ino line 832-835. But I get lost at that point not being
sure which line actually updates the "ON" or "OFF" text on the main web
page.  Still trying though.
Thanks.