arendst/Tasmota #5479: Request: add ability to calibrate white colour on RGB LED instances

dave-sausages, 2019-03-15:
I have a 30w LED floodlight which I put a Magichome module in flashed with Tasmota.

It all works fine, but the red channel appears to be brighter than the other 2 channels (which is strange given it already has the resistors in line with it from the original control board).

I can't see anyway to calibrate or adjust the red channel to reduce it and bring it inline with the blue and green ones. I can find commands to adjust the global pwm range, but not just for the one pwm channel.

ascillato, 2019-03-15:
Please, try command LEDTABLE 1 to see if that makes a better color behaviour in your led

ascillato, 2019-03-15:
Also please complete the troubleshooting template and please, post a picture of your led.

It is juat a 3 channel led ?

ascillato2, 2019-03-16:
Please, could you be so kind on completing the troubleshooting template in order to have more information so as to properly help you?

Also, please post a picture of your led. It is a 3 channel led? 

Have you tried to use the console command `ledtable 1` to see if that makes a better color behaviour in your led? 

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato2, 2019-03-17:
Closing this issue as there isn't feedback. Please, when you have time, share the requested information. Thanks.

closed 2019-03-17 by ascillato2