arendst/Tasmota #5480: HomeAssistant Commands Lost During WiFi Outage

galapogos01, 2019-03-16:
### HomeAssistant Commands Lost During WiFi Outage
I have a couple of Sonoffs flashed with Tasmota 6.4.1 and am running Hassio 0.89.1 with Mosquitto.  I enabled Auto Discovery and pretty much everything else is default (new setup).

The problem is, if HA issues a command when a switch is off the WiFi (intermittent network for example), when the Sonoff reconnects it does not change state in line with the HA command.

I understand I can try things to improve WiFi reliability (eg. currently testing Core 2.3.0), but my question is how to make the Sonoff obey the MQTT state commanded by HA when it comes back online. 
  

### REQUESTED INFORMATION

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (i.e. Sonoff Basic) : Sonoff Basic
- [X] Tasmota binary firmware version number used : 6.4.1 / (pre-compiled)
- [X] Development IDE - Compiler / Upload tools used : N/A
- [X] Provide the output of command ``status 0`` :

```
02:29:41 CMD: status 0
02:29:41 MQT: sonoff2/stat/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff Basic"],"Topic":"sonoff2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
02:29:41 MQT: sonoff2/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/020300/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:24:07","StartupUTC":"2019-03-16T01:05:34","Sleep":50,"CfgHolder":4617,"BootCount":27,"SaveCount":165,"SaveAddress":"FA000"}}
02:29:41 MQT: sonoff2/stat/STATUS2 = {"StatusFWR":{"Version":"6.4.1.22(06a3678-sonoff)","BuildDateTime":"2019-03-15T16:02:10","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
02:29:41 MQT: sonoff2/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["default",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","280500000100000000000000000000000000","00000200"]}}
02:29:41 MQT: sonoff2/stat/STATUS4 = {"StatusMEM":{"ProgramSize":507,"Free":496,"Heap":13,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
02:29:41 MQT: sonoff2/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoff2-5816","IPAddress":"192.168.0.201","Gateway":"192.168.0.5","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.5","Mac":"5C:CF:7F:A8:76:B8","Webserver":2,"WifiConfig":1}}
02:29:41 MQT: sonoff2/stat/STATUS6 = {"StatusMQT":{"MqttHost":"hassio","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A876B8","MqttUser":"removed","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
02:29:41 MQT: sonoff2/stat/STATUS7 = {"StatusTIM":{"UTC":"Sat Mar 16 01:29:41 2019","Local":"Sat Mar 16 02:29:41 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:01","Sunset":"18:56"}}
02:29:41 MQT: sonoff2/stat/STATUS10 = {"StatusSNS":{"Time":"2019-03-16T02:29:41"}}
02:29:41 MQT: sonoff2/stat/STATUS11 = {"StatusSTS":{"Time":"2019-03-16T02:29:41","Uptime":"0T00:24:07","Vcc":3.191,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"POWER":"OFF","Wifi":{"AP":1,"SSId":"default","BSSId":"removed","Channel":5,"RSSI":68,"LinkCount":1,"Downtime":"0T00:00:10"}}}
```
- [X] Provide the output of console when you experience your issue if apply : N/A
```

ascillato2, 2019-03-16:
Hi,

By default, Tasmota is configured to be in a system with a distributed control technique, meaning that the actual status of Tasmota is inside the device. So, the master of the actual status is the device itself and the rest of devices can gather that information from Tasmota without the need of storing that information in a central controller.

If you want to use a centralized control technique, where the master of all device status is a central controller like your mqtt broker, you will need to disable the auto discovery and configure all manually.

You will need to make a new entry in configuration.yaml with the config of your Tasmota device and use retain to true. 
Then, in Tasmota you will need to config powerretain to true.
There is no example of this but will be the same as the wiki explains in Home Assistant Manual Config but with retain to true. 

Remember that using that, the master always will be your broker, so if you change the state of Tasmota while it is disconnected from your mqtt, after connection, your broker will always override the Tasmota state.

We don't recommend using retained messages, but if you want a centralized approach, you have the options in Tasmota.

If you have further questions or you need further assistant, please do not hesitate on asking here or in the Tasmota support chat. Thanks

closed 2019-03-16 by ascillato2

galapogos01, 2019-03-16:
Thanks for the super prompt reply.

It's a little odd that the defaults for HA Discovery have retain: false, as this means that guaranteeing a synchronised state with the HA server or any server-side Automations requires a 100% reliable network connection.  That's not realistic.  

In my case I was finding lights left on all night because HA sent an off command that was never received, but this would be catastrophic in the case of a pump, garage door etc.

What should the defaults be to avoid noob questions or odd behaviours?

Is it possible to get a detailed description of the various "retain" configs and exactly what they do?  I found the config wiki page a little light on implications of the config.

ascillato, 2019-03-16:
Sorry, it is not odd. If you search in issues there are hundreds of issues with ghost switching and weird toggling, all because using retained messages in a wrong control scheme.
Tasmota is meant for cloudless and control distributed home automation system. So, retained messages are meant for other type of automations. If you search in the wiki at tutorials, there is a very good video from thehookup explaining that. 

If you want to increase the reliability of that the message from your HA reaches your Tasmota, you have to do other things different than retained messages. 

In Tasmota you have internal timers to turn off lights, you have pulsetime to turn off pumps, you have rules to make the behaviours you want, you have KNX to control your device without MQTT and also if you want a direct Tasmota to Tasmota communication without using your broker (HA supports KNX also), and also you can increase your wifi reliability using better routers like mikrotik.

We do not recommend retained messages. It gives more problems than solutions (search in issues) 

A wifi approach for home automation is not 100% reliable. If you want that, you need a wired system instead.

ascillato, 2019-03-16:
Sorry, it is not odd. If you search in issues there are hundreds of issues with ghost switching and weird toggling, all because using retained messages in a wrong control scheme.
Tasmota is meant for cloudless and control distributed home automation system. So, retained messages are meant for other type of automations. If you search in the wiki at tutorials, there is a very good video from thehookup explaining that.

If you want to increase the reliability that a message from your HA reaches your Tasmota, you have to do other things different than retained messages. Retained messages are not meant for syncronizing. Are meant for centralized control and to override internal status of controlled devices.

For keeping the sync, Tasmota is publishing every status changes, plus the teleperiod message.

In Tasmota you have internal timers to turn off lights, you have pulsetime to turn off pumps, you have rules to make the behaviours you want, you have KNX to control your device without MQTT and also if you want a direct Tasmota to Tasmota communication without using your broker (HA supports KNX also), and also you can increase your wifi reliability using better routers like mikrotik.

We do not recommend retained messages. It gives more problems than solutions (search in issues)

A wifi approach for home automation is not 100% reliable. If you want that, you need a wired system instead.

galapogos01, 2019-03-16:
Thanks for the reply.

I understand Tasmota's original purpose and it does that great. I guess the issue comes down to the half-support for HA; why include HA discovery if the discovered devices and config are not set up optimally and reliably?

If retained messages are not the right answer, what IS the correct configuration to ensure the HA status is honoured even when the IoT device has an unreliable connection? As stated, I am a newcomer to this ecosystem and expected the defaults to "just work".

Jason2866, 2019-03-17:
So read the wikis of Tasmota and HA. Google and search you will find a lot of stuff to learn how it works.
No Tasmota is NOT a just it works solution. If you want something like that buy a commercial Home Automation system.

galapogos01, 2019-03-17:
As I said above, I have read the wiki for hours and turned up nothing. I am politely asking for more info on what the retain settings do and help on what best practice is for the common use case of using HA with Tasmota reliably.

Your post contains no helpful info, nor answers to my questions. Why be so hostile?

Jason2866, 2019-03-17:
> Your post contains no helpful info, nor answers to my questions. Why be so hostile?

https://github.com/arendst/Sonoff-Tasmota/blob/release-6.5/CONTRIBUTING.md

galapogos01, 2019-03-18:
I posted an issue because I believe this default Discovery configuration leads to unreliability and should be addressd. See point 1 of your link.