arendst/Tasmota #5481: Streamline Template GUI in a single page

blakadder, 2019-03-16:
Would it make sense to have all the template options in the **Configure Template** menu page instead of one part being in Configure Template and the other in Configure Other.

Here's a mockup of an idea, not a perfect solution since the page has to be scrolled on certain resolutions but atleast its all in one place

[image]

You can save on space if you remove the header **Import Template** and instead write it in the submit button. Not sure it makes that option as straightforward:

[image]

arendst, 2019-03-16:
The issue is with the logic needed to upgrade the changes made to either the selects when an import is done or the template text when a select is changed; they tend to deviate this way.

blakadder, 2019-03-16:
Not worth the trouble and time then.
If the documentation is good it shouldn't be an issue having them separate as it is now

closed 2019-03-16 by arendst