arendst/Tasmota #5482: Cannot switch the Gosund SP1 smart socket over KNX

tosate, 2019-03-16:
<GUIDE>

<This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum>
  
<Please DO NOT OPEN AN ISSUE:>
  <If you have general questions or you need help on Tasmota usage, go to the Tasmota support chat>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a new device, please use the Tasmota Template Feature. See wiki for that>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### ISSUE DESCRIPTION - TROUBLESHOOTING
I have a Gosund SP1 smart socket and I want to integrate it into my KNX bus system. It is connected over a UniFi AC Lite access point from Ubiquiti Networks and it communicates with the knxd service running on a raspberry Pi. Whenever I toggle the device over the web UI, it sends its status to KNX. However, if I try to send a KNX telegram to the device nothing happens.
As far as I understand the Wiki, I need to use rules to switch the device, but I don't understand how to formulate the rule that would do the switching.

### REQUESTED INFORMATION
How can I define a rule for switching the device on and off for a specific KNX address? What other issue might prevent this from working?

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Gosund SP1
- [x] Tasmota binary firmware version number used : 6.4.1(knx) / pre-compiled
- [x] Upload tools used : tuya-convert
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
14:41:01 MQT: stat/sonoff1/STATUS = {"Status":{"Module":55,"FriendlyName":["sonoff-mq-wz"],"Topic":"sonoff1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:41:01 MQT: stat/sonoff1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"Software/System restart","Uptime":"3T22:19:17","StartupUTC":"2019-03-12T15:21:44","Sleep":0,"BootCount":15,"SaveCount":179,"SaveAddress":"F9000"}}
14:41:01 MQT: stat/sonoff1/STATUS2 = {"StatusFWR":{"Version":"6.4.1(knx)","BuildDateTime":"2018-12-24T14:49:38","Boot":4,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
14:41:01 MQT: stat/sonoff1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SaSaNet",""],"TelePeriod":300,"SetOption":["0A008009","558180C0","00000000"]}}
14:41:01 MQT: stat/sonoff1/STATUS4 = {"StatusMEM":{"ProgramSize":537,"Free":464,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","1FDAE594","000383B0","23B617CE","00003BC0"]}}
14:41:01 MQT: stat/sonoff1/STATUS5 = {"StatusNET":{"Hostname":"sonoff-wohnzimmer","IPAddress":"192.168.178.57","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"84:0D:8E:53:C2:57","Webserver":2,"WifiConfig":4}}
14:41:01 MQT: stat/sonoff1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.31","MqttPort":1883,"MqttClientMask":"DVES_LIVING","MqttClient":"DVES_LIVING","MqttUser":"home","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
14:41:01 MQT: stat/sonoff1/STATUS7 = {"StatusTIM":{"UTC":"Sat Mar 16 13:41:01 2019","Local":"Sat Mar 16 14:41:01 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:01","Sunset":"18:56"}}
14:41:01 MQT: stat/sonoff1/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
14:41:01 MQT: stat/sonoff1/STATUS10 = {"StatusSNS":{"Time":"2019-03-16T14:41:01","ENERGY":{"TotalStartTime":"2019-03-03T10:18:40","Total":0.868,"Yesterday":0.009,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
14:41:01 MQT: stat/sonoff1/STATUS11 = {"StatusSTS":{"Time":"2019-03-16T14:41:01","Uptime":"3T22:19:17","Vcc":3.532,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"OFF","Wifi":{"AP":1,"SSId":"SaSaNet","BSSId":"F0:9F:C2:74:3C:74","Channel":6,"RSSI":74}}}

```
**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-03-16:
Hi,

As requested in the Tasmota support chat, please share the screenshot of the KNX config menu.

There is no need to use rules for switching a relay over KNX. Where you have read that in the wiki? You can use directly the KNX web menu for config that.

Are you using the same group address for both transmition and reception?

ascillato, 2019-03-16:
Besides the requested information, please download to your phone the app called MULTICAST TESTER to see of the telegram you are sending to Tasmota is being broadcated.

tosate, 2019-03-16:
Here is my KNX configuration:
[image]

I can see the packets from knxd in Wireshark on my notebook which is also connected over the WiFi. However, I will also try the MULTICAST TESTER.

tosate, 2019-03-16:
Is there a MULTICAST TESTER app for iOS?

ascillato, 2019-03-16:
>Is there a MULTICAST TESTER app for iOS?

Don't know. Should be.

ascillato, 2019-03-16:
Are you sending the command to Tasmota using 1.1.112?

Please, set `weblog 4`, and then try again to send commands to Tasmota to see if there is any log in the console.

tosate, 2019-03-16:
With weblog 4 there is nothing to see.
However, I can see the package in Wireshark
[image]

ascillato, 2019-03-16:
Your Tasmota is connected directly to the main router or it has a repeater in between?

ascillato, 2019-03-16:
I still have the doubt of the wiki. In which part did you read about the need of using rules? I want to correct that.

ascillato, 2019-03-16:
Another thing we can try is to update to latest Tasmota version (from development branch) but with core 2.3.0 

http://thehackbox.org/tasmota/020300/sonoff-knx.bin

The arduino core have some issues with some brands of routers (specially with the IP multicast) 

Also we can test with core 2.5.0 

http://thehackbox.org/tasmota/020500/sonoff-knx.bin

ascillato, 2019-03-16:
Do you have any other device with esp8266 like a sonoff basic or a nodemcu?

ascillato2, 2019-03-17:
Hi, any news on this?

tosate, 2019-03-17:
The Tasmota is connected over a UniFi AC Lite access point from Ubiquiti Networks. The latter is connected to a UniFi Switch. Internet access and DHCP is provided by a Fritz!Box.

The Wiki has a KNX-Features page. It consists of an Implemented Features section. In there you can read "For using rules: receive KNX command (on/off)". Since switching the device on/off is not working, I thought that a rule is missing.

I can try other firmwares as you mentioned.

I don't have a sonoff basic or a nodemcu. I only have three other Gosund SP1 smart sockets with the same issue.

tosate, 2019-03-17:
The latest Tasmota version (from development branch) with core 2.3.0 didn't change anything.

tosate, 2019-03-17:
Same behavior with with core 2.5.0.

ascillato, 2019-03-17:
Ok. As you have other devices, try to send an on/off command directly between devices. Please, use the example of the wiki at KNX features for direct communication between Tasmotas.

ascillato, 2019-03-17:
Like 2.2.1 as relay 1 output in one device and 2.2.1 as relay 1 input in the other device, so as to check the direct communication.

tosate, 2019-03-17:
Direct communication seems to work!

----
20:51:32 KNX: Received from 1.1.113 Command Write: 1 to Output 1
20:51:32 SRC: Knx
20:51:32 KNX: Output 1 = 1 sent to 1.4.113
20:51:32 MQT: stat/sonoff2/RESULT = {"POWER":"ON"}
20:51:32 MQT: stat/sonoff2/POWER = ON
20:51:33 CFG: Saved to flash at F5, Count 72, Bytes 3584
----

ascillato, 2019-03-17:
Ok. Cool. So the issue is in the multicast configuration of your wifi setup.

You should try to modify the configuration in your routers. Seems that it is not being retransmitted the multicast telegrams over your different routers. In that case, there is much we can do from Tasmota side. 

You can try to connect your devices directly to your fritzbox.

tosate, 2019-03-17:
I just figured out that the second device is working and can be switched on/off over KNX!
Now I have to find out what is different with the first device.
They are connected to different access points. However, UniFi access points are provisioned over to UniFi controller and it should setup each AP in the same way.

Thank you very much for your support so far!

tosate, 2019-03-17:
The working device runs Program Version  6.4.1(knx) with Core/SDK Version | 2_4_2/2.2.1(cfd48f3)
How can I downgrade the other device? It's not working by providing
https://github.com/arendst/Sonoff-Tasmota/releases/download/v6.4.1/sonoff-knx.bin
as OTA URL

ascillato, 2019-03-17:
You should use minimal first.

Anyway, there is no difference in KNX driver between both versions, so I don't think downgrading will solve your issue.

If you think that is the firmware, you should try erasing all flash with esptool.py as explained in the wiki and flash again.

I suspect that the issue is just the wifi configuration of your routers.

Jason2866, 2019-03-18:
@tosate  You have 2 components which are know to be problematic (sometimes?!) with ESP8266
FritzBox and Ubiquiti. Maybe the combination of this 2 devices make things worse.
For most users a switch to core 2.3.0 solved there problems. Sad to say not for all.
We cant digg here deeper because this is closed source from espressif

ascillato2, 2019-03-18:
@tosate 

Hi, any news on this?

ascillato2, 2019-03-19:
When you have time, please, tell us if now it is working for you. Thanks

closed 2019-03-19 by ascillato2

tosate, 2019-03-19:
Sorry, I have been busy.
Unfortunately it is not working. It was working for one of the four smart sockets, but I couldn't figure out what is the difference. I keep searching.
Thanks for your help so far.