arendst/Tasmota #5483: Update 6.4.1.22 SonOff CH4 and SonOFF RF

Valerio1975, 2019-03-16:
Good evening I had a ch4 and a rf version 6.4.1.21 and the bme680 sensor on ch4 and bmp280 on RF worked perfectly. After updating to both of them at 6.4.1.22 the RF module continues to work but the CH4 module no longer works when reading the bme680 module. I checked the configurations and they are like before. Did you have another option for the sensors? or by chance in updating the file did you omit the CH4 module for the sensors?
Would it be possible to have the 6.4.21 sensor files so I try to reload this?

ascillato, 2019-03-16:
If your issue is solved, please close it. Thanks

closed 2019-03-16 by ascillato2

Valerio1975, 2019-03-16:
ok i thought you were interested to know that it doesn't restart properly after restarting the upgrade.

ascillato2, 2019-03-16:
Ah, ok. Thanks for reporting. Yes, some sensors require power cycle to start working again.

ascillato2, 2019-03-16:
Please, next time could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)