arendst/Tasmota #5485: How to extract the AnalogA0 input value from Wemos D1 mini MQTT topic and change value in Home Assistant

garret, 2019-03-17:
I installed latest version of Tasmota on a Wemos D1 mini and also commented the ADC voice to read the AnalogA0 input value. This value is given correctly in the main page ranging from 0 to 1024. 

```
09:14:01 MQT: tele/wemos_3/STATE = {"Time":"2019-03-17T09:14:01","Uptime":"0T00:40:20","SleepMode":"Dynamic","Sleep":50,"LoadAvg":38,"POWER":"OFF","Wifi":{"AP":1,"SSId":"In_medio_stat_virtus-2GHz","BSSId":"A0:63:91:E0:A4:87","Channel":1,"RSSI":98}}
09:14:01 MQT: tele/wemos_3/SENSOR = {"Time":"2019-03-17T09:14:01","ANALOG":{"A0":1024}}
```

When I go in the console I can see that every 5 minutes a new MQTT topic is fired with the Analog value. I would like to take this value, report it in Home Assistant and also convert it. 

However, first I did not understand how to extract this value to be reported in Home Assistant. This is the code I am using which is for sure not correct. 

```
sensor:
  - platform: mqtt
    name: "Plant Humidity"
    state_topic: "tele/wemos_3/SENSOR"
    unit_of_measurement: '%'
    value_template: "{{ value_json.A0 }}"
```

Another question would be if you know how to convert directly in home assistant this value ranging from 0 to 100% knowing that 1024 would be 100%. I know it can be done but I am very newbie to this. 

Finally, the analog input is attached to a soil moistuire sensor. Attached to the same Wemos there is a relay shield. I noticed that when I switch on/off the relay, also the analog value is a little bit influenced (like it changes to 1018 from 1024 for instance). Is that normal?

blakadder, 2019-03-17:
Should be `value_json.ANALOG.A0` since A0 is nested under analog
the other question is for the home assistant support, it involves jinja templating and is out of the tasmota realm

ascillato2, 2019-03-17:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-03-17 by ascillato2

orlandperz, 2019-04-25:
A thousand apologies for my English. I had the same concern and resolved it as follows:

`value_template: "{{(value_json ['ANALOG']. A0 | float * 100/1024) | round (1)}}"`

So I can convert an analogous reading into a value for example percentage in a simple way without having to use jinja templating.

gbananachewicz, 2022-10-05:
I have mqtt client on Android. 
From mqtt i get
{"Time":"2022-10-05T12:32:15","ANALOG":{"Range":1178}}

And all the same I see in app
[image]

How can I see only value - 1178?

barbudor, 2022-10-05:
That's a problem  on configuring your app, not a Tasmota problem
We don't know your app and how it is working. Ask the support team of your app