arendst/Tasmota #5486: Counter Problem

AingTii, 2019-03-17:
I tried to use the Counter function.

I set GPIO15 to Counter1n and i set the debounce to 3200.
on GPIO15 is a reedcontact for the gas Counter.

The Problem is:
the Counter couts sometimes if the reed contact Close, sometimes if open, sometimes both states and sometimes it counte more than +1.
And it doesn't matter what time between the states..

So how can I solve that Problem or is that a bug in tasmota?

ascillato2, 2019-03-17:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

netpok, 2019-03-18:
You probably want to use Counter 1, Counter 1n is the same but without pull-up in that case you have to provide your own pull-up resistor. Generally you want to use Counter 1, Counter 1n is for some special use-cases.

ascillato2, 2019-03-18:
Closing this issue as there is no feedback. Please, when you have time, complete the troubleshooting template. Thanks.

closed 2019-03-18 by ascillato2

AingTii, 2019-03-18:
okay