arendst/Tasmota #5487: remove user selectable gpio assignments from shelly

blakadder, 2019-03-17:
People are universally dumb and we had a too many (imho) inquiries about attaching peripherals to Shelly1 exposed gpios even after being strongly advised against that and pointing them to the wiki warning. 

**Describe the solution you'd like**
Completely remove user configurable gpio's for shelly products. If someone really knows what they're doing they'll figure it out with templates instead of having one dumbass electrocuting himself or someone else.

underwoodblog, 2019-03-17:
Didn't agree. Because two things are eternal: The universe and human stupidity. So a "dumbass" finds a soluten to electrocute himself anyway. 

**Describe the solution you'd like**
Let Darwin sort them all out.
https://teespring.com/shop/let-darwin-sort-them-all-out#pid=212&cid=5822&sid=front

meingraham, 2019-03-17:
Yes, remove configurable GPIOs from the built-in modules.

But still leave a warning that if they decide to configure a Shelly via a template (e.g., using Shelly as the BASE), they need to beware. Maybe amp up the warning to say modifications of Shelly devices are **highly NOT recommended** unless they **absolutely know what they are doing** and that any modifications they perform, like any device, is completely their responsibility, etc., etc., etc.

The alternative is that they see that the device has exposed GPIOs and they make modifications and have not been given a warning. They go forward with what looks similar to other devices and end up harming themselves! Better that they are provided with information and warnings than to hope inexperienced users don't delve into making modifications.

ascillato, 2019-03-17:
Totally agree with @Blakadder
That is a safer path.
The proof that this is needed are the insistent questions for adding sensors to a non isolated device in the Tasmota support chat.

ascillato2, 2019-03-18:
PR https://github.com/arendst/Sonoff-Tasmota/pull/5489

PR #5489 (2019-03-19): Hiding User Configurable Pins for SHELLY 1
Some users had tried to connect sensors to shelly 1, being it powered by mains AC. So, as it is a very dangerous condition and a safety concern, those extra GPIO are being taken from the template. Comments about this, were left on the code for people who really know what they are doing.

## Description:

**Related issue (if applicable):** fixes #5487

## Checklist:
  - [x] The pull request is done against the dev branch
  - [x] Only relevant files were touched (Also beware if your editor has auto-formatting feature enabled)
  - [x] Only one feature/fi [... cut]
closed 2019-03-18 by ascillato2