arendst/Tasmota #5488: Calibration Dictionary

KDMcMullan, 2019-03-17:
**Is your feature request related to a problem? Please describe.**
Due to manufacturing intolerances, DS18B20 temerature sensors do not all read the same. I have read several different specs. Some manufacturers claim +/-0.5degC accuracy. Others claim only +/-2.05degC accuracy. I'm monitoring temperatures at different positions in the in a hot water tank. Because the differences are subtle, I'd like to know I'm getting better accuracy than 2 degC.

**Describe the solution you'd like**
Despite the above, I have actually found the sensors themselves to be extremely accurate, just not very well calibrated. After software calibration, they are really accurate. Would it be possible, please, to add linear calibration values to temperatures sensors? This would involve TASMOTA storing a dictionary entry for each sensor containing a scaling factor and offset. TASMOTA would multiply by the scaling factor (default 1) and add the offset (default 0), yielding up an extremely accurate temperature sensor, regardless the manufacturer.

**Have you look for this feature in other issues and in the wiki?**
Different from https://github.com/arendst/Sonoff-Tasmota/issues/2248 in that this includes a little more research and an implementation suggestion.

**Describe alternatives you've considered**
The scale and offset have been successfully applied in the end user application. This can be done, for example, using the templating feature in Home Assistant, but I feel that this is a setting which belongs closer tot eh device, such that other clients can benefit from the calibration, rather than each client performing its own.

**Additional context**
I'd be glad to have a stab at this myself, but might need some coaching. I'm a software engineer, but with only limited exposure to C.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2019-03-17:
Hi, 

Your idea is good and logical, but it will require more use of flash and RAM, and as memory is a very small resource in the esp8266, I don't think is going to be added soon.

This has been requested before a couple of times and the solution provided with the actual features of Tasmota is the one explained in the wiki at: https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#8b-adjust-the-value-of-a-sensor-and-send-it-by-mqtt

Hope this fits your requirements.

KDMcMullan, 2019-03-17:
Thank you.

I'm using a prebuilt binary at present. Does this say that I need to build my own executable to enable rules? Assuming so, where do I actually putt he rules?

NB I thought we were OKay for flash as long a anyone wanting to update remembered to flash a minimal version first?

Thanks again.

ascillato, 2019-03-17:
Using the precompiled bins, rules feature are already available.
To write any rule you want, you can type them in the console.

ascillato2, 2019-03-18:
Closing this issue as it has been answered.

Added the label Feature Request (Hold Over) for future reference. Thanks for sharing your ideas. If you have further questions/ideas/etc, please just ask here or in the Tasmota Support Chat. Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-03-18 by ascillato2