arendst/Tasmota #5490: RGB remapping improvements

netpok, 2019-03-18:
In relation to #5326 there were a new request for another color combination which is not supported. We came up with two solution:

- Implement it as a new device or
- To solve these kind of problems once and for all (the famous last words) introduce a new remapping system to cover the currently known configurations, by using a full 6 (5 color + not used) to 12 channel (dual my92x1) remapping

I support the latter but it would lead to the generalizations of AiLight and other modules it also requires 4 bytes (3 if we count the SetOption37 which will be freed) of EEPROM space, so I'm requesting additional opinions (@arendst, @gsimon75).

dlashua, 2019-03-18:
In the referenced Issue (#5326) I don't see that @DavinKD has 12 channels. He has the same bulb I do from what I can tell. 

I've seen two configurations used by device makers so far. A single My9291. And a Double My9231. That would be 4 channels in the first case, and 6 channels (total, 3 per MY9231) in the second case.

In theory, there could be many more MY9231s daisy chained together, but, from what I've seen, device manufacturers aren't even using all of the channels provided by 2 MY9231s. To future proof, yes, 9 or 12 is a good idea. But to support what it out there now, 6 would be fine. 

Dealing with multiple same colored channels complicates matters further. It moves the configuration storage space requirements from (if "n" is the number of channels) n! to n^n (or, in the case of 6 channels, from 720 to 46656). 

In the same way that the PWM1-5i channels/Pins are not typically used on devices with MY92X1 chips, the MY92X1 channels are not typically used on devices that use PWM. There are 5 allocatable PWM Pins and, with devices I've seen today, this is enough to cover them. Not knowing the Tasmota internals, I don't know if it's possible... but perhaps coming up with a more generic way to support storing the configuration for "This device has RED LEDs" that would cover PWM and MY92X1 would save some space?

Devices need 5 logical outputs: R, G, B, Wc, Ww. Devices have up to 6 physical outputs. These can be PWM, PWMi, MY92X1, and even LED, and LEDi depending on how generic the code gets.

That seems like a lot of work, though, for one, potentially, small device class. Making a new device type seems more simple. But, as we've seen, SetOption37 became needed to support all the different configurations one device type might have. A problem that PWM didn't have because each pin was assigned individually.

netpok, 2019-03-18:
~~The 12 channel was brought to my attention by @gsimon75 in [this comment](https://github.com/arendst/Sonoff-Tasmota/pull/5326#issuecomment-473793745) so that's all I know.~~

Misunderstanding here, it's actually two different 6 channel mapping, so the 6^6 would suffice and that only requires 2bytes.

gsimon75, 2019-03-18:
My mistake, I haven't paid enough attention to the actual code in xdrv_04_light.ino

I've jumped to wrong conclusions because of the fact that we initalise that whole `duty[][]` array, even though we always use only one row of it.

There's indeed only one active set of channels, and it's indeed only 6, not 12, so the config possibilities are 6 ^ 6  = 46656, so it requires only two bytes, and you are right :).

digiblur, 2019-03-18:
In the bulbs I've seen with the MY92xx drivers I've seen the normal RGB and RGBW types, but then you have some odd balls like the Lohas I've covered in some videos that have a 4 channel chip with it done like CW,W,CW,W interlaced, then another with RGB,CW,CW.  I'm sure there are many other oddballs out there.

DavinKD, 2019-03-18:
Digiblur has covered my only oddity which is a Lohas bulb with 2 CW channels.

farfarfromhome, 2019-04-11:
I have found another oddity.  I have a Lohas 6" downlight RGB&CW.  It has two MY92X1 chips and three banks of CW LEDs.  I used Digiblur's ESPHome method and determined CW is on channels 0, 1 and 2, and RGB on channels 5, 4 and 3 respectively.

closed 2019-06-18 by stale[bot]