arendst/Tasmota #5492: MQTT commands for display

sydknee, 2019-03-18:
Please delete, laugh or whatever you choose.

Fairly new to Tasmota and so far loving it.

I'm just getting into adding a display to my project (currently one of them little OLeds) some of the modes don't appear to work very well but I understand this is still in development.

Anyway stop the waffling and get on with the question.  Can Display commands be sent over MQTT?  I've tried a few to no effect .  I'd like to be able to switch between modes ie displays the date time then flicks to text displays text sent via MQTT then flicks back to date time.

Date and time take up the top half of the screen in Mode 1, Mode 3 and 5 seem to work but don't display the time.  Mode 1 is OK but only uses half the screen.

Any info not in the documentation would be very useful, doesn't seem to be many people posting on the internet about displays.

ascillato, 2019-03-18:
All Tasmota commands that you write in the console can be sent also by mqtt.

Jason2866, 2019-03-18:
Maybe this doc is a help https://github.com/arendst/Sonoff-Tasmota/files/2365181/DisplayText.docx

ascillato2, 2019-03-18:
Closing this issue as it has been answered.

Please, if you need further assistant, please find us in the Tasmota support chat. Thanks 

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-03-18 by ascillato2