arendst/Tasmota #5493: Ability to have multiple WS281X lights each with individual Schemes

robin13, 2019-03-18:
I would like to do an implementation with multiple strings of WS2812 lights.  Specifically I would like to have one string (connected to D2/GPIO4) with 60 lights using Scheme 5 (clock), and another string of 10 lights (connected to D4/GPIO2) using Scheme 1 (fixed color).
From the current command structure, I'm not sure how it would be possible to address multiple lights connected to a single board, but it would be pretty neat if it was possible. :)

Jason2866, 2019-03-18:
Not possible with Tasmota. 
Will not implemented because to much resources needed

ascillato2, 2019-03-18:
Closing this issue as it has been answered.

Thanks for sharing your ideas. 

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-03-18 by ascillato2