arendst/Tasmota #5494: Serial data reception using SERIAL_7E1

Bry50, 2019-03-18:
Describe the solution you'd like

I would like to connect my connected energy counter (linky) to Tasmota. 
Linky Serial data is sent using: 1200 bps, 7 data bits, even parity, 1 stop bit

Describe alternatives you've considered

I have downloaded and flashed the latest Tasmota Master.
Program Version	6.3.0
Build Date & Time	2019-03-18T15:43:36
Core/SDK Version	2_3_0/1.5.3(aec24ac9)

I am trying to use Hardware Serial Bridge
Sonoff is configured as 
23 Sonoff 4CH PRO
All three GPIO are set to 00 None

Using protocol 8N1, I can send and receive data from termite ( baudrate 9600).

In issue [#3837](https://github.com/arendst/Sonoff-Tasmota/issues/3837), it is recommended :
You will have to change SERIAL_8N1 for SERIAL_7E1 in the TasmotaSerial.cpp

```
bool TasmotaSerial::begin(long speed) {
if (m_hardserial) {
Serial.flush();
Serial.begin(speed, SERIAL_8N1);
```

TasmotaSerial.cpp was slightly different, so I inserted
```
bool TasmotaSerial::begin(long speed, int stop_bits) {
  m_stop_bits = ((stop_bits -1) &1) +1;
  if (m_hardserial) {
    Serial.flush();
    if (2 == m_stop_bits) {
      Serial.begin(speed, SERIAL_7E2);
    } else {
      Serial.begin(speed, SERIAL_7E1);
    }
```

However, using 7E1 on termite
From Sonoff; Serialsend data is received by Termite OK
but
From Termite, data is not correctly received by the Sonoff
Often, no data is received, occasionally the last one or two characters are received.

Surprisingly, Using 8N1 on termite the serial data is still received correctly.

ascillato2, 2019-03-18:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Also, please update to latest Tasmota (6.4.1.22). Your version 6.3 is very old and lots of changes related to serial were implemented, so there won't be support over an old version. 

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

Jason2866, 2019-03-18:
7E1 is not supported

Frogmore42, 2019-03-18:
Since I have a need for 9-bit serial, I am working on extending the support.  What exact requirements are there?

Bry50, 2019-03-18:
Thank you for the reply.
Sorry for not correctly filling in the template. 
I am in no way a programmer and it will take me some time to  download and try the newer version ( I am not on the computer a lot).
The French enegy meter uses 9600 or 1200 bps, 7 data bits, even parity, 1 stop bit,

Once I got to read the serial information ( which requires an optocoupler link) I was hoping to send the recieved serial information to Home Assistant via MQTT and pharse out the power consumption information.

Even flashing a Sonoff is daunting for the inexperienced, changing code is almost unthinkable.
It would be desirable if, just like changing the baud rate, a command could select the desired serial protocol.

I was thinking of raising an enhancement request if got the link to work.

Frogmore42, 2019-03-23:
The current code for serial bridge does not appear to use HW serial.  The default for Tasmota Serial is software only, so the change you made does nothing.  This appears to be a bug in serial bridge.

Bry50, 2019-03-25:
Thanks.
I tried configuring the GPIO2 to 71 SerBr TX and connecting the serial interface to SDA, but it seemed to make my Sonoff unstable, but I have been having a lot of network problems and I'm not sure if the instability is due to this.  

So before trying further, do you thing the software serial bridge might work ?

Frogmore42, 2019-03-26:
Software serial is unlikely to be successful (long term at least).  It seems interrupts and WiFi don't play well together.
https://github.com/espressif/ESP8266_NONOS_SDK/issues/90

Also, the current software serial library doesn't support 7-bit nor does it support even parity.

Now, it does look like there was an intention to support HW Serial with the serial bridge.

Change this line:
`SerialBridgeSerial = new TasmotaSerial(pin[GPIO_SBR_RX], pin[GPIO_SBR_TX]);`
to
`SerialBridgeSerial = new TasmotaSerial(pin[GPIO_SBR_RX], pin[GPIO_SBR_TX], 1);`

That will tell it to use Serial1 for serial.  You will have to make sure you assign the GPIOs correctly.  Your change to SERIAL_7E1 is also needed.  That should get data flowing.

Bry50, 2019-03-27:
Thanks very much indeed......
7E1 ( 7 data bits, Even parity, 1 stop bit) now works in both transmit and receive modes 
(at least for 1200 and 9600 bps)

For info, here's what I did
AAA
In xdrv_08_serial_bridge.ino ; changed
SerialBridgeSerial = new TasmotaSerial(pin[GPIO_SBR_RX], pin[GPIO_SBR_TX]);
to
SerialBridgeSerial = new TasmotaSerial(pin[GPIO_SBR_RX], pin[GPIO_SBR_TX], 1);

BBB
Did a global find SERIAL_8N1 and replace with SERIAL_7E1
( not sure if it was required to change everywhere, but I thought why not..)

CCC
In my_user-config.h; I commented out a lot of sensors I will never use.

DDD
In the Sonoff conffiguration
Configured module type to Sonoff 4CH Pro (23)
Configured GPIO1 to Serial TX (148)
Configured GPIO3 to Serial RX (149)

Thanks again, 
I now have to check cables, test MQTT to home assistant 
and decipher the incoming message, but that is another chapter....

This issue can be closed.

Frogmore42, 2019-03-27:
Actually, there is an issue here.  You probably shouldn't have needed to make the change in xdr_08_serial_bridge.ino

That probably should default to use serial 1 hardware when you choose those gpio pins.

Jason2866, 2019-03-28:
Please close this issue since it is answered. THX

Bry50, 2019-03-28:
Hi
After many, many tests 
( after all,  I am a nooby who a little time ago though flashing an overwhelming task)
I have found that that the following is sufficient to use protocol 7E1

In the Sonoff conffiguration
Configured module type to Sonoff 4CH Pro (23)
Configured GPIO1 to Serial TX (148)
Configured GPIO3 to Serial RX (149)

In  TasmotaSerial.cpp ; change to
```
bool TasmotaSerial::begin(long speed, int stop_bits) {
  m_stop_bits = ((stop_bits -1) &1) +1;
  if (m_hardserial) {
    Serial.flush();
    if (2 == m_stop_bits) {
      Serial.begin(speed, SERIAL_7E1);
    } else {
      Serial.begin(speed, SERIAL_7E1);
    }
```
But **also** in sonoff.ini; change to 
`SerialConfig serial_config = SERIAL_7E1;`

I think it would be a plus to have a Command to set the serial interface protocol,
do you think it worthwhile to raise an enhancement request?
If it is too complicated, then it is not probably worth it....

Is it for me to close the issue? If so, I'm not sure how to do so

Jason2866, 2019-03-28:
Great, you find a solution! 
Yes please close this issue.

Bry50, 2019-04-03:
Sorry to reopen this issue, but with the changes I made, the baudrate seems to be fixed at 9600. 
When i try to change the baudrate to 1200 and I issue the command Baudrate 1200, it just replies Baudrate 9600.
The french energy meter has two modes which uses both baudrates. 
I can live with 9600, but it complicates my testing. 
If anyone can point me in the direction of a fix, That would be great. If not, I'll reclose the issue.

Frogmore42, 2019-04-03:
Try the software serial baudrate command.

Or fix this part of the code in sonoff.ino
1080    else if (CMND_BAUDRATE == command_code) { 
1081       if (payload32 > 1200) { 
1082         payload32 /= 1200;  // Make it a valid baudrate 
1083         baudrate = (1 == payload) ? APP_BAUDRATE : payload32 * 1200; 
1084         SetSerialBaudrate(baudrate); 
1085       } 
1086       Response_P(S_JSON_COMMAND_NVALUE, command, Settings.baudrate * 1200); 
1087     } 

Line 1081 needs to use >=

Bry50, 2019-04-06:
sorry for the delay in replying ( I was travelling)
Magic; that worked
Thanks again

closed 2019-04-06 by Bry50