arendst/Tasmota #5495: MQTT set retain optional for tele INFO(1|2|3)

panzerdev, 2019-03-19:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**

Yes. As I want to switch my devices to DHCP and not static IPs anymore, I'd like to have a way to find out what IPs the devices got via MQTT. 
There are the INFO1, INFO2 and INFO3 messages published already, but they don't have a retain flag. Without that flag if you miss the publish of that value It's gone til the next restart.

**Describe the solution you'd like**

https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/my_user_config.h#L234
This flag aready exists and could be added to the publishing of the INFO(n) msg.

https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xdrv_02_mqtt.ino#L317
https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xdrv_02_mqtt.ino#L322
https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xdrv_02_mqtt.ino#L327

**Describe alternatives you've considered**

A publish to a topic could result in those infos being published again. This would lead to a bit of a request response pattern that I am not a fan of with mqtt tbh.

**Additional context**

No 

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-03-19:
I dont use Hass, so i dont know how it works... 
Isnt it possible just to query the info you want (via mqtt) and update to the actual values in your HA?
This is easy and straight forward. Retain isnt always correct...

panzerdev, 2019-03-19:
- What do you mean by "Hass"?

- I couldn't find a MQTT command to get the IP for example. In the code I see the publishing of those values being done in `MqttConnected()` I would assume that these values only get published when MQTT gets connected.

- And yes retained data is not always right.. but it's better then nothing.

blakadder, 2019-03-19:
Command IpAddress1 without a value returns current ip address of the device

arendst, 2019-03-19:
As does command ``status 5``

panzerdev, 2019-03-19:
Thanks @arendst for the hint with the `status 5` command. I am not the biggest fan of this request response type of thing but it totally get's the job done. 

I thought as the flag is called `MQTT_TELE_RETAIN`, is used for the `tele` topic and the `INFO(n)` are published on that topic, it would be a streamlined thing to make them optionally retained as well. If not I think we can close that issue here as resolved :smiley:

digiblur, 2019-03-19:
There are many ways to to do this, you could have an automation periodically send an ipaddress command to the sonoffs group topic, you could write a rule to have each device send it's state every so often, etc.

Jason2866, 2019-03-19:
@panzerdev  HASS (short for Home Assistant) normally users from this solution are asking for retain...
You opened this issue you can close it :-)

closed 2019-03-19 by panzerdev