arendst/Tasmota #5499: Can't turn off Power1 blink

JetDriver, 2019-03-19:
<GUIDE>

<This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum>
  
<Please DO NOT OPEN AN ISSUE:>
  <If you have general questions or you need help on Tasmota usage, go to the Tasmota support chat>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a new device, please use the Tasmota Template Feature. See wiki for that>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### ISSUE DESCRIPTION - TROUBLESHOOTING
_A clear description of what the issue is and be as extensive as possible_
  After trying out "Power<x> 4" to see how it worked, I can't turn it off. I need it for relays 2 and 3 (a DPDT latching relay module) and it's working great there. But I don't want blink on Relay1, the ac output. I have upgraded to Tasmota 6.5.1 and that had no effect. Any suggestions?

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : TH16_____
- [x ] Tasmota binary firmware version number used : 2_3_0/1.5.3(aec24ac9)____ / (pre-compiled or self-compiled ?)
- [x ] Development IDE - Compiler / Upload tools used : _OTA___ / ____
- [x ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
22:21:11 CMD: status 0
22:21:11 MQT: stat/sonoff/STATUS = {"Status":{"Module":4,"FriendlyName":["Greenhouse","Sonoff2","Sonoff3","Sonoff4"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:21:11 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:26:09","StartupUTC":"2019-03-19T20:55:02","Sleep":50,"CfgHolder":4617,"BootCount":30,"SaveCount":150,"SaveAddress":"F4000"}}
22:21:11 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-classic)","BuildDateTime":"2019-03-19T12:25:42","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
22:21:11 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["Emily","indebuurt2"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00000009","280500000100000000000000000000000000","00000001"]}}
22:21:11 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":484,"Free":516,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","6010A390","008403A8","210217C6","000000C0"]}}
22:21:11 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-7205","IPAddress":"192.168.0.185","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"DC:4F:22:0D:DC:25","Webserver":2,"WifiConfig":3}}
22:21:11 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.162","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0DDC25","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:21:11 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Tue Mar 19 21:21:11 2019","Local":"Tue Mar 19 22:21:11 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00"}}
22:21:11 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-03-19T22:21:11","AM2301":{"Temperature":34.5,"Humidity":5.0},"TempUnit":"C"}}
22:21:11 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-03-19T22:21:11","Uptime":"0T00:26:09","Vcc":3.151,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"Emily","BSSId":"50:C7:BF:F3:7F:9B","Channel":2,"RSSI":62,"LinkCount":1,"Downtime":"0T00:00:06"}}

```
- [x ] Provide the output of console when you experience your issue if apply :

_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
22:26:07 CMD: power1 4
22:26:07 MQT: stat/sonoff/RESULT = {"POWER1":"Blink OFF"}
22:26:25 CMD: power1 1
22:26:25 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
22:26:25 MQT: stat/sonoff/POWER1 = ON
22:26:26 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
22:26:26 MQT: stat/sonoff/POWER1 = OFF

```
**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-03-20:
Please use weblog 4 for more debug information

JetDriver, 2019-03-20:
Here's console output with weblog 4:

02:56:28 CMD: power1 1
02:56:28 SRC: WebConsole from 192.168.0.142
02:56:28 RSL: Received Topic /power1, Data Size 1, Data 1
02:56:28 RSL: Group 0, Index 1, Command POWER, Data 1
02:56:28 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
02:56:28 MQT: stat/sonoff/POWER1 = ON
02:56:28 SRC: PulseTimer
02:56:28 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
02:56:28 MQT: stat/sonoff/POWER1 = OFF

Thanks

ascillato, 2019-03-20:
Ok, so it is not the command blink. You have set the command pulsetimer.

Go to the console and type `pulsetimer1 0` to disable the previous config you made.

JetDriver, 2019-03-20:
Mil gracias! That did the trick, although my console was dead and I had to do a "Reset 6" to get back to zero. Sos argentino, no? Mi esposa tambien. Saludos!

closed 2019-03-20 by JetDriver