arendst/Tasmota #5500: IRrecv no longer works since version 6.4.0

MeisterAdebar, 2019-03-19:
With the 6.4.0 and 6.5.0 i constantly get the following messages, I do not do anything.

22:47:21 MQT: tele/tasmotair/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":3,"Data":"0x24AE7D4C"}}
22:47:22 MQT: tele/tasmotair/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":3,"Data":"0x22AE7A2A"}}
22:47:23 MQT: tele/tasmotair/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":7,"Data":"0x1372AF3A"}}
22:47:24 MQT: tele/tasmotair/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":3,"Data":"0x24AE7D4C"}}
22:47:25 MQT: tele/tasmotair/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":4,"Data":"0x69DBBC3D"}}
22:47:26 MQT: tele/tasmotair/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":4,"Data":"0x6A19A4D0"}}
22:47:29 MQT: tele/tasmotair/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":4,"Data":"0x6A19A4D0"}}
22:47:31 MQT: tele/tasmotair/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":12,"Data":"0x8A1A9624"}}
22:47:33 MQT: tele/tasmotair/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":4,"Data":"0x6A19A4D0"}}
22:47:35 MQT: tele/tasmotair/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":4,"Data":"0xE9E70E8C"}}
22:47:36 MQT: tele/tasmotair/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":4,"Data":"0x44172A6A"}}
22:47:38 MQT: tele/tasmotair/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":7,"Data":"0x6FF93866"}}
22:47:41 MQT: tele/tasmotair/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":5,"Data":"0xACDF9408"}}
22:47:41 MQT: tele/tasmotair/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":3,"Data":"0x22AE7A2A"}}
22:47:42 MQT: tele/tasmotair/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":3,"Data":"0x22AE7A29"}}
22:47:43 MQT: tele/tasmotair/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":3,"Data":"0x24AE7D4C"}}
22:47:44 MQT: tele/tasmotair/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":4,"Data":"0x8E1C1C14"}}
22:47:45 MQT: tele/tasmotair/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":3,"Data":"0x24AE7D4C"}}
22:47:45 MQT: tele/tasmotair/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":4,"Data":"0x6A19A4D0"}}

ascillato, 2019-03-20:
It is giving you the DATA field. It is working. What are you expecting?

ascillato2, 2019-03-20:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

Jason2866, 2019-03-20:
Just tested it works with latest release 6.5.0.

ascillato2, 2019-03-21:
Please, when you have time, complete the troubleshooting template in order to have more information so as to properly help you. Thanks.

closed 2019-03-21 by ascillato2

mills1975, 2019-05-13:
Hello!
I have a similar problem after updating my Basic from 6.2.1 to 6.5.
While IRRecv worked like a charm in 6.2.1, now after updating to 6.5, my IRRcv constantly receives odd
IR-Codes. (dont know from where they come)
07:31:19 MQT: tele/sonoff_marshall/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":3,"Data":"0x25AE7EE0"}}
07:31:20 MQT: tele/sonoff_marshall/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":5,"Data":"0x9A2D8247"}}
07:31:23 MQT: tele/sonoff_marshall/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":9,"Data":"0x4468DD6F"}}
07:31:25 MQT: tele/sonoff_marshall/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":10,"Data":"0x5D4C6DF"}}
07:31:27 MQT: tele/sonoff_marshall/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":5,"Data":"0x1838D170"}}
07:31:29 MQT: tele/sonoff_marshall/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":3,"Data":"0xFEAC02E5"}}
07:31:30 MQT: tele/sonoff_marshall/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":4,"Data":"0x6A19A4D3"}}
07:31:32 MQT: tele/sonoff_marshall/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":4,"Data":"0xC1E490FF"}}
07:31:34 MQT: tele/sonoff_marshall/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":9,"Data":"0x709F0C8C"}}
07:31:36 MQT: tele/sonoff_marshall/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":5,"Data":"0xA51F01A0"}}
07:31:39 MQT: tele/sonoff_marshall/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":7,"Data":"0x3D12D9F6"}}
07:31:41 MQT: tele/sonoff_marshall/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":9,"Data":"0x3A9D19AB"}}
07:31:44 MQT: tele/sonoff_marshall/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":3,"Data":"0x25AE7EE0"}}

TBR-BRD, 2021-05-26:
I have the same Problem with the Version 9.4.0, do you have an idea?