arendst/Tasmota #5501: Make communication methods selectable for WS2812 devices

nst197, 2019-03-20:
There seems to be a lot of WS2812 led strips around that don't work with the default bit banging method. You currently have to compile your own version of Tasmota to use Uart or enable DMA mode. Would it be possible to make the communication method for WS2812 devices selectable from the configuration menu i.e. DMA / Uart / Bit banging and speeds 400/800? 

I'm thinking another drop down next to the device type might work, and could be filtered to only show the methods that work for each GPIO pin i.e Uart only shows for GPIO2, and DMA only shows for GPIO3.

ascillato2, 2019-03-20:
Closing this issue as it is **duplicated**. Sorry.

It can't be done due the buffers are different and everything can't fit in the esp8266. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-03-20 by ascillato2

nst197, 2019-03-20:
@ascillato2 I looked for other requests about this before I logged it but couldn't find any. Can you point me in the right direction please so I can understand what the issues are a bit better?